f4exb/sdrangel #2312: Cannot transmit microphone input using SDRAngel, USB sound card and HackRF One

DerrickLau, 2024-11-05:
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

derrick@TARDIS:~$ hackrf_info
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc287c23c3
Board ID Number: 4 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x004f4f66
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

Bus 002 Device 009: ID 0c76:1695 JMTek, LLC. USB PnP Audio Device
  bDeviceClass            0 
  bDeviceProtocol         0 
  iProduct                1 USB PnP Audio Device
      (Bus Powered)
Couldn't open device, some information will be missing

SDRangel v 7.22.2-74-gb122d40bc

Opened Tx:0 device as HackRF.  Frequency 435 MHz.  Checked RF Amp.  Created T:0 channel.  Added NFM Modulator.  Pressed the Microphone button.
Went to SDRangel->Preferences->Audio->Audio Input.  Selected device also_input.usb-0c76_USB_PnP_Audio_Device-00.mono-fallback.

Talked into microphone but only hear static on my receiver (BaoFeng UV-9R handheld).  Using the standard dipole antenna included with HackRF One.

Please advise.

srcejon, 2024-11-06:
Would suggesting post a screenshot showing the TX spectrum (and all other settings).

DerrickLau, 2024-11-06:
Screenshots of audio preferences and TX spectrum (including Tx device, channel, WFM modulator).
[image]
[image]

DerrickLau, 2024-11-06:
John Telek had the same issue and believes it is a FIFO queue issue with the audio input.  Here is his debug output:

2024-11-07 09:03:29.534 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:03:37.114 (W) libpng warning: iCCP: known incorrect sRGB profile
2024-11-07 09:03:37.114 (W) libpng warning: iCCP: too many profiles
2024-11-07 09:04:34.587 (W) QObject: Cannot create children for a parent that is in a different thread.
(Parent is ChannelAPI(0x56098103cc78), parent's thread is QThread(0x56097f240fd0), current thread is QThread(0x560980003370)
2024-11-07 09:04:34.589 (W) SampleSourceFifo::write: underrun (write too slow) using 512000 old samples
2024-11-07 09:04:34.591 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:34.591 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:38.607 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:41.057 (W) QObject: Cannot create children for a parent that is in a different thread.
(Parent is ChannelAPI(0x56098103cc78), parent's thread is QThread(0x56097f240fd0), current thread is QThread(0x560980003370)
2024-11-07 09:04:41.059 (W) SampleSourceFifo::write: underrun (write too slow) using 512000 old samples
2024-11-07 09:04:41.060 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:41.060 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:44.875 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:47.367 (W) QObject: Cannot create children for a parent that is in a different thread.
(Parent is ChannelAPI(0x56098103cc78), parent's thread is QThread(0x56097f240fd0), current thread is QThread(0x560980003370)
2024-11-07 09:04:47.369 (W) SampleSourceFifo::write: underrun (write too slow) using 512000 old samples
2024-11-07 09:04:47.370 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:47.371 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:52.184 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:54.700 (W) QObject: Cannot create children for a parent that is in a different thread.
(Parent is ChannelAPI(0x56098103cc78), parent's thread is QThread(0x56097f240fd0), current thread is QThread(0x560980003370)
2024-11-07 09:04:54.701 (W) SampleSourceFifo::write: underrun (write too slow) using 512000 old samples
2024-11-07 09:04:54.703 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found
2024-11-07 09:04:54.703 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x774470044700 not found

f4exb, 2024-11-07:
You have to click on the microphone icon of the NFM modulator in order to activate the audio input

drom, 2024-11-24:
Looks like I have the same issue related to #2336

srcejon, 2024-11-25:
Closed in favour of #2336