f4exb/sdrangel #2317: 7.22.3 crashes in Mac OS Sequoia 15.1

Axpelle, 2024-11-10:
Using Airspy Mini in frequency scanner mode, when the start/stop scanner button is pressed and then the Start/Stop Acquisition is pressed SDRAngel crashes. I'm using a Mac mini M2. Receiver is WDSP.

srcejon, 2024-11-10:
Can you post a picture of your settings please

Axpelle, 2024-11-11:
> Can you post a picture of your settings please

Screenshots of settings as requested. In fact SDRa crashes as soon as I hit the "start acquisition" button.
[Settings 1.pdf](https://github.com/user-attachments/files/17694333/Settings.1.pdf)
[Settings 2.pdf](https://github.com/user-attachments/files/17694335/Settings.2.pdf)
[Settings 3.pdf](https://github.com/user-attachments/files/17694336/Settings.3.pdf)

Axpelle, 2024-11-11:
And I add that SDRa 7.22.2 for Intel (running on Rosetta) does not crash when I hit the "start acquisition" button.

srcejon, 2024-11-12:
Closing in favour of the other issue

closed 2024-11-12 by srcejon