f4exb/sdrangel #2318: ADS-B demodulation map no longer zoomable (flatpak Qt 6 problem)

CocolinoFan, 2024-11-10:
Hello,

The last update introduced some problems (I am using the Flatpak version of SDRangel):

- Can no longer zoom in and out in the ADS-B demodulator map.
- ADS-B demodulator is missing the `mapboxgl` Map provider (which is not great as they have this "Street" map that is really good).

Version of SDRangel with this features working: `SDRangel 7.22.2 Qt 5.15.10 OpenGL 4.6 x86_64 KDE Flatpak`
Version of SDRangel with this not features working: `SDRangel 7.22.2 Qt 6.8.0 OpenGL 4.6 x86_64 KDE Flatpak`

AsciiWolf, 2024-11-10:
This is most likely related to the new build using Qt6 instead of Qt5.

srcejon, 2024-11-10:
> This is most likely related to the new build using Qt6 instead of Qt5.

Yes - mapboxgl not yet supported in Qt6. This wasn't a problem for Windows moving to Qt6 as it didn't work anyway - and there were other benefits for Windows (E.g. better DPI scaling). The .deb and snaps for Linux are still built using Qt 5.

CocolinoFan, 2025-01-04:
No please, don't close the issue Mr. github-actions the problem is **not fixed**.
From my understanding the project moved to Qt6.
The move made SDRangel better on Windows but broke the Linux version. A clear regression. Here is the comment: https://github.com/f4exb/sdrangel/issues/2318#issuecomment-2466725557
The "solution" presented there is to use an older version of SDRangel that still uses Qt5. I hope that was an out of season April fools joke.
I really hope the project is not planning to abandon Linux.

srcejon, 2025-01-04:
> The "solution" presented is to use an older version of SDRangel that still uses Qt5. I hope that was an out of season April fools joke. 

The workaround was to use a current version of sdrangle that uses qt5.

CocolinoFan, 2025-01-04:
I am a very smooth brained individual. That's why I like flatpak, it always works.

> The workaround was to use a current version of sdrangel that uses qt5.

As far as I can tell SDRangel is not in the Ubuntu repositories.
```
cocolino@Fucputer:~/Downloads$ sudo apt install sdrangel
Error: Unable to locate package sdrangel
```

Tring to install SDRangel .deb file from the release packages gives me:
```
cocolino@Fucputer:~/Downloads$ sudo apt install /home/cocolino/Downloads/sdrangel_7.22.5_ubuntu-24.04_amd64.deb 
Note, selecting 'sdrangel' instead of '/home/cocolino/Downloads/sdrangel_7.22.5_ubuntu-24.04_amd64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

Unsatisfied dependencies:
 sdrangel : Depends: libavcodec60 (>= 7:6.0) but it is not installable
            Depends: libavformat60 (>= 7:6.0) but it is not installable
            Depends: libavutil58 (>= 7:6.0) but it is not installable
            Depends: libswresample4 (>= 7:6.0) but it is not installable
            Depends: libswscale7 (>= 7:6.0) but it is not installable
            Depends: libuhd4.6.0t64 (>= 4.6.0.0+ds1) but it is not installable
            Depends: qtbase-abi-5-15-13 but it is not installable
Error: Unable to correct problems, you have held broken packages.
```
```
cocolino@Fucputer:~/Downloads$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.10
Release:        24.10
Codename:       oracular
```
Yes, the package is for Ubuntu 24.04 and I have Ubuntu 24.10 but I am not going to reinstall and downgrade my operating system.

I can wait if the problem will be fixed. I just though that, for Linux, >=Qt6 breaks functionality, and it is what it is, tough luck Linux users.

AsciiWolf, 2025-01-04:
If there are bigger problems with the current Flatpak using Qt6, I can revert the "runtime" version used by this Flatpak back to Qt5 (KDE runtime 5.15 instead of 6.8), but this is not a good long-term solution. And from my testing, the Qt6 build of SDRangel did work pretty well, except some smaller issues like this one.

closed 2025-09-25 by CocolinoFan