f4exb/sdrangel #2322: MacOs Intel version request

loewal, 2024-11-11:
Please compile an Intel MacOS version as well, because ARM versions don't recognize SDRplay devices like RSP1a.

Thank you!

srcejon, 2024-11-11:
There's an x64 build here you can try:

https://github.com/srcejon/sdrangel/actions/runs/11765840554

The latest arm version should support SDRplay. If you look in the logs:

It downloads the API 

    wget https://www.sdrplay.com/software/SDRplayAPI-macos-installer-universal-3.15.0.pkg

and it finds the libraries:

    -- SDRPLAY LIBRARIES /usr/local/lib/libsdrplay_api.dylib
    -- SDRPLAY INCLUDE DIRS /usr/local/include
    -- Found SDRPLAY: /usr/local/lib/libsdrplay_api.dylib

What do you see in your log file when starting? Do you have API 3.15 installed?

loewal, 2024-11-11:
Installed the api 3.15 and restarted my Mc Mini M2: no sdrplay.
Installed 7.22.4 Intel. I took almost 2 minutes to get past "loading plugins"...

How can I start and see the logfiles?

loewal, 2024-11-11:
I have the same situation with SDR++. Arm-version finds no SDRplay, even after installation of API.
I was somehow afraid to do that, because in the past even the Intel-version didn't recognize the RSP1a anymore. Now it still does with SDR++ Intel.

srcejon, 2024-11-11:
To save a log file, enable it under Preferences > Logging.

loewal, 2024-11-12:
But how can a logfile be useful, when I cannot select an SDRplay device a the source?

loewal, 2024-11-12:
This is what I get when using the ARM-version:
2024-11-12 07:28:23.389 (W) PluginManager::loadPluginsDir: Cannot load library /Applications/Korte_Golf/SDRangel ARM.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib: dlopen(/Applications/Korte_Golf/SDRangel ARM.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib, 0x0085): Library not loaded: libsdrplay_api.so.3
  Referenced from: <3FA5415D-5F1E-377B-89A2-08A2C3BBDC92> /Applications/Korte_Golf/SDRangel ARM.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib
  Reason: tried: 'libsdrplay_api.so.3' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibsdrplay_api.so.3' (no such file), 'libsdrplay_api.so.3' (no such file), '//libsdrplay_api.so.3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS//libsdrplay_api.so.3' (no such file), '//libsdrplay_api.so.3' (no such file)
2024-11-12 07:28:23.389 (I) PluginManager::loadPluginsDir: Soapy SDR disabled skipping libinputsoapysdr.dylib
2

srcejon, 2024-11-12:
What does `ls -l /usr/local/lib` show for you?

Is libsdrplay_api.so.3 is that directory? If not, where is it?

loewal, 2024-11-12:
Yes it is, but on several places. I have to do a cleanup.
Thanks anyway.

loewal, 2024-11-12:
Cleaned Soapy and api references.
Installed Api 3.15 and looked for libsdrplay_api.so.3.

ls -l /usr/local/lib/libsdrplay_api.so.3

lrwxr-xr-x  1 root  admin  37 12 nov 09:25 /usr/local/lib/libsdrplay_api.so.3 -> /usr/local/lib/libsdrplay_api.so.3.15

Can you make something of it?
No support in the ARM version. Intel runs fine...

loewal, 2024-11-12:
Addendum:

2024-11-12 15:45:28.362 (W) PluginManager::loadPluginsDir: Cannot load library /Applications/Korte_Golf/SDRangel 2.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib: dlopen(/Applications/Korte_Golf/SDRangel 2.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib, 0x0085): Library not loaded: libsdrplay_api.so.3
  Referenced from: <3FA5415D-5F1E-377B-89A2-08A2C3BBDC92> /Applications/Korte_Golf/SDRangel 2.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib
  Reason: tried: 'libsdrplay_api.so.3' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibsdrplay_api.so.3' (no such file), 'libsdrplay_api.so.3' (no such file), '//libsdrplay_api.so.3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS//libsdrplay_api.so.3' (no such file), '//libsdrplay_api.so.3' (no such file)

srcejon, 2024-11-12:
> lrwxr-xr-x 1 root admin 37 12 nov 09:25 /usr/local/lib/libsdrplay_api.so.3 -> /usr/local/lib/libsdrplay_api.so.3.15

Ok, that looks good.

> libinputsdrplayv3.dylib, 0x0085): Library not loaded: libsdrplay_api.so.3

The problem seems to be it's looking for libsdrplay_api.so.3, but doesn't say where. You could possibly set export DYLD_LIBRARY_PATH=/usr/local/lib - but not sure that is really well supported on Mac anymore.

Currently, we are changing libsdrplay_api.so.3.15 to /usr/local/lib/libsdrplay_api.so.3.15 - but I guess that needs to now be libsdrplay_api.so.3 to /usr/local/lib/libsdrplay_api.so.3. So you might be able to fix it with: 

     install_name_tool -change  libsdrplay_api.so.3 /usr/local/lib/libsdrplay_api.so.3  /Applications/Korte_Golf/SDRangel 2.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib

Need to check what x86 version does.

PR #2327 (2024-11-12): SDRplayv3: Patch path to libsdrplay_api.so.3
Mac build now seems to link to libsdrplay_api.so.3 rather than libsdrplay_api.so.3.15. So patch path to that as well. Should fix #2322. I don't know if there's a better way to do it.
closed 2024-11-12 by f4exb