f4exb/sdrangel #2324: 7.22.4 crashes, when starting WDSP on MacOS

thomasgi1, 2024-11-11:
When starting a configuration containing the WDSP plugin, SDRangel immediately crashes:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SDRangel [67773]
Path:                  /Applications/SDR/SDRangel.app/Contents/MacOS/sdrangel
Identifier:            SDRangel
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-11-11 22:35:54.3112 +0100
OS Version:            macOS 12.7.6 (21H1320)
Report Version:        12
Anonymous UUID:        00D0AAE4-4FA2-1AA2-1DC3-429829F4143B

Sleep/Wake UUID:       0842925F-4BBD-4D18-A97D-B1A0559FDE66

Time Awake Since Boot: 510000 seconds
Time Since Wake:       8978 seconds

System Integrity Protection: enabled

Crashed Thread:        17  QThread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000700003463ff8
Exception Codes:       0x0000000000000002, 0x0000700003463ff8
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [67773]

VM Region Info: 0x700003463ff8 is in 0x700003463000-0x700003464000;  bytes after start: 4088  bytes before end: 7
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      Stack                    7000033e1000-700003463000 [  520K] rw-/rwx SM=PRV  thread 5
--->  STACK GUARD              700003463000-700003464000 [    4K] ---/rwx SM=NUL  ...for thread 17
      Stack                    700003464000-7000034e6000 [  520K] rw-/rwx SM=PRV  thread 17

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff8022d393a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff8022d3ca8 mach_msg + 56
2   CoreFoundation                	    0x7ff8023d729d __CFRunLoopServiceMachPort + 319
3   CoreFoundation                	    0x7ff8023d5928 __CFRunLoopRun + 1276
4   CoreFoundation                	    0x7ff8023d4d6c CFRunLoopRunSpecific + 562
5   HIToolbox                     	    0x7ff80b0875e6 RunCurrentEventLoopInMode + 292
6   HIToolbox                     	    0x7ff80b08734a ReceiveNextEventCommon + 594
7   HIToolbox                     	    0x7ff80b0870e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
8   AppKit                        	    0x7ff804e13aa9 _DPSNextEvent + 927
9   AppKit                        	    0x7ff804e12166 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
10  AppKit                        	    0x7ff804e04818 -[NSApplication run] + 586
11  libqcocoa.dylib               	       0x1070049dd 0x106fec000 + 100829
12  QtCore                        	       0x10ac293f6 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 486
13  QtCore                        	       0x10ac1ffc7 QCoreApplication::exec() + 119
14  SDRangel                      	       0x105f37148 main + 3816
15  dyld                          	       0x11031352e start + 462

Thread 1:
0   libsystem_pthread.dylib       	    0x7ff80230bf48 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	    0x7ff80230bf48 start_wqthread + 0

Thread 3:: QThread
0   libsystem_kernel.dylib        	    0x7ff8022da06a poll + 10
1   QtCore                        	       0x10adb3890 qt_safe_poll(pollfd*, unsigned int, QDeadlineTimer) + 64
[... 543 lines cut ...]
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "5f72dc58705eff005a46b3a9",
      "factorPackIds" : {

      },
      "deploymentId" : 240000015
    },
    {
      "rolloutId" : "60356660bbe37970735c5624",
      "factorPackIds" : {

      },
      "deploymentId" : 240000027
    }
  ],
  "experiments" : [

  ]
}
}
```

It seems that the stack is being corrupted.

Axpelle, 2024-11-12:
I can confirm the above. No crash when using AM Demodulator on airband instead of WDSP

sofakng, 2024-11-19:
I'm new to SDR but I'm trying to use my RTL-SDR with the nightly of SDRAngel and it crashes after I select any preset (ADS-B, ISS, etc).

srcejon, 2024-11-20:
> I'm new to SDR but I'm trying to use my RTL-SDR with the nightly of SDRAngel and it crashes after I select any preset (ADS-B, ISS, etc).

Make sure the RTL SDR isn't running when you try selecting one of the presets.

DanTheDane, 2025-02-21:
I'm experiencing the same issue, only in WDSP, where it crashes as soon as I start acquisition (in this case with a RTL-SDR v3) is enabled. I'm running MacOS 15.3.1

Axpelle, 2025-02-27:
I believe the author of SDRA is aware of this issue. Pending a fix I have reverted to using the AM demodulator when listening to airband.

srcejon, 2025-02-27:
thread #20, name = 'QThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x17c4dfff8)
  * frame #0: 0x00000001915f213c libsystem_pthread.dylib`___chkstk_darwin + 60
    frame #1: 0x000000014c3e38d8 libwdsp.dylib`WDSP::EMNR::G::G(int, double, int, std::__1::vector<double, std::__1::allocator<double>>&, std::__1::vector<float, std::__1::allocator<float>> const&) + 44
    frame #2: 0x000000014c3e5d78 libwdsp.dylib`WDSP::EMNR::calc() + 1848
    frame #3: 0x000000014c3e6090 libwdsp.dylib`WDSP::EMNR::EMNR(int, int, int, float*, float*, int, int, int, int, double, int, int, int) + 144
    frame #4: 0x000000014c3fcfa8 libwdsp.dylib`WDSP::RXA::RXA(int, int, int, int) + 2064
    frame #5: 0x000000014bb2a10c libwdsprx.dylib`WDSPRxSink::WDSPRxSink() + 412
    frame #6: 0x000000014bb2d948 libwdsprx.dylib`WDSPRxBaseband::WDSPRxBaseband() + 100
    frame #7: 0x000000014bb1e61c libwdsprx.dylib`WDSPRx::start() + 244

borjam, 2025-03-20:
So that you know, confirmed also for 7.22.6 on MacOS 15.3.2 if I enable WDSP and I activate the SDR.

My receiver is a SDRPlay RSPdx-R2 but I think it is irrelevant.

thomasgi1, 2025-03-20:
The bug was already fixed 3 weeks ago, but unfortunately there is no official release yet containing the bug fix. Until an official release will be avaliable you can use my private builds from here: [thomasgi/sdrangel/SDRangel Mac release build #3](https://github.com/thomasgi1/sdrangel/actions/runs/13823328342).

Axpelle, 2025-03-21:
> The bug was already fixed 3 weeks ago, but unfortunately there is no official release yet containing the bug fix. Until an official release will be avaliable you can use my private builds from here: [...]

Thank you so much for the fix. I can confirm that WDSP works fine in this release. Not sure if it beats the AM demodulator on airband but otherwise it's performing well. I've also noticed that optimising the squelch setting is a bit tricky in AM mode but I can live with that. Thanks once again.

thomasgi1, 2025-03-21:
> Thank you so much for the fix

I did not make the fix by myself. It was provided by [srcejon](https://github.com/srcejon). I only forked the repo to run the GitHub actions by myself in order to get the DMGs.

ianks, 2025-03-22:
@srcejon your patch is causing issues for me still, but may be an entirely separate thing

```
Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libqcocoa.dylib               	       0x107528080 0x1074ec000 + 245888
1   QtGui                         	       0x109256a84 QWindowPrivate::setVisible(bool) + 752
2   QtWidgets                     	       0x1055acdb8 0x105540000 + 445880
3   QtWidgets                     	       0x1055948cc QWidgetPrivate::hide_sys() + 312
4   QtWidgets                     	       0x105596be0 QWidgetPrivate::hide_helper() + 144
5   QtWidgets                     	       0x105597b20 QWidgetPrivate::setVisible(bool) + 960
6   QtWidgets                     	       0x10559752c QWidget::setVisible(bool) + 708
7   QtWidgets                     	       0x10576b7f8 QDialogPrivate::setVisible(bool) + 320
8   QtWidgets                     	       0x10579c47c QProgressDialog::reset() + 48
9   QtWidgets                     	       0x10579c73c QProgressDialog::setValue(int) + 528
10  libsdrgui.dylib               	       0x1052a0254 LoadConfigurationFSM::restoreGeometry() + 664
11  QtCore                        	       0x1070819d4 0x106fb4000 + 842196
12  QtStateMachine                	       0x1051f9fac 0x1051f4000 + 24492
13  QtStateMachine                	       0x10520c288 QStateMachinePrivate::enterStates(QEvent*, QList<QAbstractState*> const&, QList<QAbstractState*> const&, QSet<QAbstractState*> const&, QHash<QAbstractState*, QList<QPropertyAssignment>>&, QList<QAbstractAnimation*> const&) + 780
14  QtStateMachine                	       0x1052089f0 QStateMachinePrivate::microstep(QEvent*, QList<QAbstractTransition*> const&, CalculationCache*) + 344
15  QtStateMachine                	       0x105210688 QStateMachinePrivate::_q_process() + 236
16  QtStateMachine                	       0x1052133b4 QStateMachinePrivate::handleTransitionSignal(QObject*, int, void**) + 672
17  QtStateMachine                	       0x1051f9e20 0x1051f4000 + 24096
18  QtCore                        	       0x1070819a4 0x106fb4000 + 842148
19  QtCore                        	       0x1070819d4 0x106fb4000 + 842196
20  QtCore                        	       0x106fb96b8 0x106fb4000 + 22200
21  QtCore                        	       0x107079e10 QObject::event(QEvent*) + 88
22  QtWidgets                     	       0x10554c94c QApplicationPrivate::notify_helper(QObject*, QEvent*) + 272
23  QtWidgets                     	       0x10554d7a0 QApplication::notify(QObject*, QEvent*) + 504
24  QtCore                        	       0x107036208 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
25  QtCore                        	       0x1071aaa14 QTimerInfoList::activateTimers() + 1096
26  libqcocoa.dylib               	       0x107500cdc 0x1074ec000 + 85212
27  CoreFoundation                	       0x1848b48a4 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
28  CoreFoundation                	       0x1848b4838 __CFRunLoopDoSource0 + 176
29  CoreFoundation                	       0x1848b459c __CFRunLoopDoSources0 + 244
30  CoreFoundation                	       0x1848b3138 __CFRunLoopRun + 840
31  CoreFoundation                	       0x1848b2734 CFRunLoopRunSpecific + 588
32  HIToolbox                     	       0x18fe21530 RunCurrentEventLoopInMode + 292
33  HIToolbox                     	       0x18fe27348 ReceiveNextEventCommon + 676
34  HIToolbox                     	       0x18fe27508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
35  AppKit                        	       0x18842a848 _DPSNextEvent + 660
36  AppKit                        	       0x188d90c24 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
37  AppKit                        	       0x18841d874 -[NSApplication run] + 480
38  libqcocoa.dylib               	       0x107501938 0x1074ec000 + 88376
39  QtCore                        	       0x10703fd14 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 540
40  QtCore                        	       0x10703685c QCoreApplication::exec() + 112
41  sdrangel                      	       0x104dc31ac main + 2832
42  dyld                          	       0x18444c274 start + 2840

Thread 1:: QThread
0   libsystem_kernel.dylib        	       0x184793830 poll + 8
1   QtCore                        	       0x1071a9828 qt_safe_poll(pollfd*, unsigned int, QDeadlineTimer) + 60
2   QtCore                        	       0x1071ade24 QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 904
3   QtCore                        	       0x10703fd14 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 540
4   QtCore                        	       0x10711ba24 QThread::exec() + 284
5   QtCore                        	       0x1071ababc 0x106fb4000 + 2063036
6   libsystem_pthread.dylib       	       0x1847cc2e4 _pthread_start + 136
7   libsystem_pthread.dylib       	       0x1847c70fc thread_start + 8
```

UncleBuck62, 2026-01-12:
macOS Tahoe 26.1 with SDRAngel 7.22.10 still appears to have this issue.
I also have revered to AM Demod until this problem can be addressed

srcejon, 2026-01-14:
I just updated to Tahao 26.2 with 7.23.0 and WDSP seems to run without crashing for me, so need some more data...

3ProFilm, 2026-10-09:
Correction: my crash is a different bug, not this macOS stack overflow. On 7.27.2 Windows, WDSP Rx itself runs fine; SDRangel only crashes when the channel report is requested via the REST API (`GET .../channel/{n}/report`), because `WDSPRx::webapiFormatChannelReport()` writes to a null `getSsbDemodReport()`. I'm opening a separate issue for it, as suggested above.