f4exb/sdrangel #2329: 7.22.4 on Ubuntu opens like on Android

alphafox02, 2024-11-13:
First, thank you for continuing to support Ubuntu 22.04 and nice to see you're able to build releases on github.

I just grabbed the latest build, upgraded over my previous install, and upon launching SDRAngel it opens/presents itself as it does on Android. What I mean by that is that two small screens open, one being the "configurations" menu where it's asking what config do you want to load. The second window is the main SDRAngel I'm used to opening, but it's tiny for some reason. 

Is this the new expected behavior across all platforms? Just wanting to make sure it's not a bug.

srcejon, 2024-11-13:
It will open the Configurations window if there's no default configuration (i.e. the first time it runs).

The main window shouldn't be tiny though. There isn't any intentional code for that, as far as I know.

alphafox02, 2024-11-13:
After closing and reopening the main window opens at the size I believe I closed it at, which was much larger. 

Only on the initial opening was the main window nearly the same if not a little smaller than the config window. Seems fine otherwise.

closed 2024-11-13 by alphafox02