f4exb/sdrangel #2333: Error building 'demodadsb'

gvanem, 2024-11-15:
Using MSVC on Win-10 to build the **demodadsb** plugin, causes this error:
```c
adsbdemodsinkworker.h(54): error C2079: 'ADSBDemodSinkWorker::MsgConfigureADSBDemodSinkWorker::m_settings' uses undefined struct 'ADSBDemodSettings' 
adsbdemodsinkworker.h(79): error C2079: 'ADSBDemodSinkWorker::m_settings' uses undefined struct 'ADSBDemodSettings'
adsbdemodsinkworker.h(44): error C2440: 'return': cannot convert from 'const int' to ' const ADSBDemodSettings &' 
...
```

But with this change it now builds for me:
```diff
--- a/adsbdemodsinkworker.h 2024-11-15 09:16:23
+++ b/adsbdemodsinkworker.h 2024-11-15 10:39:28
@@ -26,6 +26,7 @@
 #include "util/message.h"
 #include "util/messagequeue.h"
 #include "adsbdemodstats.h"
+#include "adsbdemodsettings.h"

 class ADSBDemodSink;
-struct ADSBDemodSettings;
-struct ADSBDemodStats;
```

srcejon, 2024-11-15:
> Using MSVC 
What version?

gvanem, 2024-11-15:
`Version 19.42.34430 for x64`. The latest.

PR #2334 (2024-11-16): ADSB: Include adsbdemodsettings.h
Fixes #2333
closed 2024-11-16 by f4exb