f4exb/sdrangel #2336: SSB Modulator connection to microphone

drom, 2024-11-17:
SDRangel: v7.22.4, v7.22.3, v7.22.2, v7.22.1
OS: Linux OpenSuse Tumbleweed
SDR Device: PlutoSDR

All work great on RX.
On TX + SSB modulator: Connecting any Audio input source. SSB modulator shows -150dB and don't show any movement. 

[image]

All was working fine on version v7.21.3

drom, 2024-11-20:
I am using pipewire . Can it be related? Should I try pulseaudio?

srcejon, 2024-11-21:
> I am using pipewire . Can it be related? Should I try pulseaudio?

Worth a try.

drom, 2024-11-21:
did not helped :(

drom, 2024-11-24:
Switching between modulators I could get WFM Modulator working once. But All other (Audio sourced) modulators don't work. NFM, AM, SSB

drom, 2024-11-24:
When adding SSB Modulator I am getting this errors:
```
2024-11-24 22:06:17.683 (W) libpng warning: iCCP: known incorrect sRGB profile
2024-11-24 22:06:17.683 (W) libpng warning: iCCP: too many profiles
2024-11-24 22:06:17.693 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x562ba446c678 not found
2024-11-24 22:06:17.693 (W) AudioDeviceManager::removeAudioSource: audio FIFO 0x562ba446c678 not found
```

srcejon, 2024-11-25:
On Windows, it's working with Qt6, but not working with Qt5.

It goes wrong, I think, because AMModBaseband::applySettings calls AMModSource::applyChannelSettings setting channelSampleRate to 0. Then AMModBaseband::handleData is called, and the thread locks up.

With Qt6, although the channelSampleRate is set to 0, before AMModBaseband::handleData is called, AMModBaseband::handleMessage: gets a DSPSignalNotification with the correct channel sample rate.

Why is channelSampleRate set to 0? Because m_channelizer->setBasebandSampleRate is only set when AMModBaseband gets a DSPSignalNotification, and it currently never gets one. This is because AMMod gets the DSPSignalNotification while not running, and it only forwards if running.

What I suspect we should do is send a DSPSignalNotification notification in AMMod::start(). That is what all of the demods do. Eg in AMDemod::start:

    m_thread->start();

    DSPSignalNotification *dspMsg = new DSPSignalNotification(m_basebandSampleRate, m_centerFrequency);
    m_basebandSink->getInputMessageQueue()->push(dspMsg);

    AMDemodBaseband::MsgConfigureAMDemodBaseband *msg = AMDemodBaseband::MsgConfigureAMDemodBaseband::create(m_settings, true);
    m_basebandSink->getInputMessageQueue()->push(msg);

    m_running = true;

Unrelated, but m_feedbackAudioSampleRate is uninitialised:
 
    2024-11-25 13:23:48.148 (W) AMModSource::applyFeedbackAudioSampleRate: invalid sample rate -842150451

jamesa5771, 2024-11-28:
I have had the same issue too with audio inputs on SSB it will also not output CW but will just about do a tone but seems to be  tempramental

drom, 2024-11-28:
@srcejon can we cut a release that I can try?

MihaiRF, 2024-12-06:
Hello. I have the same problem when trying to use the microphone input in the SSB modulator. It worked before, now it doesn't.
Configuration: Linux Mint 22 (Ubuntu 24.04), HackRF card, SDRangel 7.22.5

drom, 2024-12-10:
The issue exists in v7.22.5

srcejon, 2024-12-10:
Just tried again, and while 7.22.5 is fixed for File Output, Test Output and Audio Output, it's not working on USRP (don't have a Pluto or HackRF), so a problem still remains.

srcejon, 2024-12-10:
Looks like there's another bug as well, introduced by the Sonar Fixes [patch](https://github.com/srcejon/sdrangel/commit/b85419c56a119a8b07be7c512b0ba0c5cd614b68#diff-9e3bd5bb03b3511f3f58ee79bca3a7ece3b66bbcbc37d08e9daed9a9e3ed86c7R128), when m_channelSampleRate is greater than m_audioSampleRate. Easy fix though.

closed 2024-12-10 by f4exb

drom, 2024-12-11:
Lets hope for new release soon :crossed_fingers:

PR #2357 (2024-12-10): Fix audio modulators when channel sample rate is greater than audio sample rate
Should fix #2336.

(Also added a couple of small DSC changes).