f4exb/sdrangel #2339: Sdrangel crashes when decoding FT8 cross 6K bandwidth

Tony-QQ, 2024-11-20:
I’m a newbie of SDR and love to use Sdrangel.

There are some FT8 bands like 7.071MHz and 7.074MHz which can be covered by one FT8 demodulator with 6K bandwidth. But Sdrangel crashes in a few seconds or minutes if set the FT8 demodulator to 6K bandwidth.

I use the Windows version of Sdrangel 7.22.4 on windows11 home edition, and tested by Kiwisdr and RSP1, both can reproduce this issue. But it is strange that if reduce FT8 decoding bandwidth from 6K to 5.6K, Sdrangel doesn’t crash.

Thanks.

f4exb, 2024-11-20:
Something fishy here although one would never use such a large bandwidth but as it is proposed...
```
#0  0x00007fa790143883 in FT8::FT8::reduce_rate (this=0x7fa73001a620, a=..., hz0=4040.06665, hz1=6000, arate=<optimized out>, brate=<optimized out>, delta_hz=@0x7fa7029fd340: 0) at /opt/build/sdrangel/ft8/ft8.cpp:446
446                 bbins[i] = bins1[i + delta];
(gdb) p i
$1 = 31280
(gdb) p delta
$3 = 59176
(gdb) p bbins
$2 = std::vector of length 32251, capacity 32251 = ...
(gdb) p bins1
$4 = std::vector of length 90001, capacity 90001 = ...
```
31280 + 59176 = 90456 so `i + delta` is out of bounds.

Tony-QQ, 2024-11-21:
Thank you so much for taking a look on this issue.

Sdrangel has a super good FT8 performance and user friendly interface especially for multi-band FT8. I’m listening on 10m-160m FT8 band almost 7/24 by Sdrangel. There’re 17 frequency slots from 10m-160m，but my web SDR device limit to only 13 channels output. So 6K FT8 bandwidth helps a lot to let me cover more FT8 frequency like at 20m/30m/40m. 

[image]

Maybe this is a rare use case, but hope to make  Sdrangel be better and better. ^^

f4exb, 2024-11-21:
It is true that in some cases the 3k bands are adjacent

f4exb, 2024-12-01:
I don't really understand the algorithm but it seems that for the upper bound of the filter we cannot go as close to the internal sample rate (12k) divided by two (6k). There must be a guard of ~200 Hz. So by limiting the upper bound to 5.8k this should fix the problem.

Tony-QQ, 2024-12-02:
Many thanks for the fix!  
5.8K is good enough for coverage of wideband ft8 decoding.

f4exb, 2024-12-02:
Now in commit [ec4d14b](https://github.com/f4exb/sdrangel/commit/ec4d14be3f44f7c0cb5cafe8195d8e0a8c30ce56)
 ·

f4exb, 2024-12-10:
Deployed in v7.22.5

closed 2024-12-10 by f4exb