f4exb/sdrangel #2343: Windows audio not working in SDRAngel 7.22.4

DerrickLau, 2024-12-02:
Have Windows 10 Pro on Dell laptop with RealTek sound card and headset with microphone plugged into mini-plug port.  Installed SDRAngel 7.22.4.  Set Audio input to RealTek sound card headset microphone in Preferences > Audio.  Created Tx device for HackRF One.  Added NFM modulator channel.  Clicked on microphone device.  Pressed play.  BaoFeng receiver heard static, but when I spoke into my headset mike, no voice transmitted and no motion on the waterfall spectrum display in SDR Angel.  Note I am able to transmit my voice using SDRAngel 7.22.0 on customized Ubuntu named DragonOS.

srcejon, 2024-12-02:
Duplicate of https://github.com/f4exb/sdrangel/issues/2336

srcejon, 2024-12-02:
If you want to try a test build that has a fix: https://github.com/f4exb/sdrangel/actions/runs/12118897964/artifacts/2261390124

closed 2025-01-17 by github-actions[bot]