f4exb/sdrangel #2348: The microphone has no sound (with sound feedback turned on) when using SSB modulation

RealCommonCat, 2024-12-03:
The microphone works fine when WFM is used, but there is no sound when SSB is used (with sound feedback turned on).
At the same time the sound signal is not transmitted but the" tone modulation "is normal

[image]
[image]
[image]

I also found that the older version (7.15.4) does not have this problem

srcejon, 2024-12-03:
Duplicate of https://github.com/f4exb/sdrangel/issues/2336

RealCommonCat, 2024-12-03:
OK

closed 2024-12-03 by RealCommonCat