f4exb/sdrangel #2349: channel analyzer channel scope displays old buffer data

PlasmaHH, 2024-12-03:
Running sdrangel 7.22.4 from openSUSE TW with nvidia graphics I have the problem that the channel scope does not display data properly, it flickers between what seems to be at least two older buffer contents. When I move other windows over it I get ghost images of those windows. 

Here is a screenshot that somehow shows whats going on:
[image]

All other applications and games using the GPU run fine here.

PlasmaHH, 2025-01-20:
So whats this mean? This works as expected?

PlasmaHH, 2025-05-06:
> I am experiencing the same issue:
> 
> 
> * Framework 13 AMD laptop
> * Debian 12 with KDE Wayland
> * SDRangel 7.22.6-8-gba673bf3a

Since this issue was close my assumption is that this is working as expected, I have no idea though whats the purpose... I don't know anyone where it isn't the same as for me, so my guess it that its a feature I don't understand.

Zorbatron, 2025-06-23:
This seems to be a wayland specific issue because this doesn't happen when running SDRAngel with `QT_QPA_PLATFORM=xcb sdrangel`. 
This forces QT to use it's X11 backend (xwayland for me). I am on release `7.22.7`

Abdull, 2025-08-11:
I can happily report that this ghost image issue does no longer manifest in my new setup.

I guess the issue got resolved because of any ***or a combination*** of the following changes to my setup:

* up-to-date build of [sdrangel *master* branch, commit `5ae8ab0bf3f4d95645d2f7f0abdc08f00dbb5509`](https://github.com/f4exb/sdrangel/commit/5ae8ab0bf3f4d95645d2f7f0abdc08f00dbb5509), which incorporates [#2498](https://github.com/f4exb/sdrangel/pull/2498). `./sdrangel --version : SDRangel 7.22.8-12-g5ae8ab0bf`
* I removed self-built mesa/vulkan/amdgpu/DRM/GLES/GLX stuff from `/usr/local/lib` ***and*** instead use the latest distribution-provided stuff for my system, i.e. everything that Debian 12 [*mesa-vulkan-drivers/**bookworm-backports***](https://packages.debian.org/bookworm-backports/mesa-vulkan-drivers) pulls in and thereby brings Mesa to version *25.0.7-2~bpo12+1* (as well as updates to other graphics-related stuff to more recent versions compared to Debian 12 "original bookworm".