f4exb/sdrangel #2350: Cannot open PlutoSDR

vatprcyq, 2024-12-04:
PlutoSDR on Windows 11
IIO and Serial present in devmgmt..
but SDRangel can't sample device
Login to plutosdr via serial is OK and password unchanged.
[sdrangel1205.log](https://github.com/user-attachments/files/18013016/sdrangel1205.log)

f4exb, 2024-12-05:
Please check the documentation: https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/plutosdrinput 

> ☞ When running the Pluto on Ethernet interface you have to create a non discoverable device reference in the [user arguments dialog](https://github.com/f4exb/sdrangel/blob/master/sdrgui/deviceuserargs.md) from the main window Preferences > Devices menu. You must use the PlutoSDR hardware ID then specify the device address with a uri parameter in the user arguments for example: uri=ip:192.168.1.10. Note that this will become effective once SDRangel is restarted.

f4exb, 2024-12-05:
For some reason it does not find the serial number:
```
2024-12-05 01:51:43.055 (C) PlutoSDRInput::openDevice: open serial TBD failed
```

vatprcyq, 2024-12-05:
[image]

vatprcyq, 2024-12-05:
SDR++ tested OK for this

vatprcyq, 2024-12-05:
Try iio_info -s :
[image]
Output:
iio_info -s
Unable to create Local IIO context : Function not implemented (40)
Available contexts:
        0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial= [usb:2.2.5]
And try copy add uri=usb:2.2.5 at user arguments can mitigate this problem.
[image]
After add this,seems OK now...
Log Attached.
[sdrangel.log](https://github.com/user-attachments/files/18019339/sdrangel.log)

F5OEO, 2025-01-17:
This serial issue comes probably from a chinese clone which could not read the qspi NOR serial number.

closed 2025-03-05 by github-actions[bot]