f4exb/sdrangel #2354: RSPDx receiver not visible in receiver list

jcfpi, 2024-12-08:
With SDRAngel  version 7.22.5 (sdrangel-7.22.5-win64.exe) : SDRplay RSPDx receiver not visible in receiver list

srcejon, 2024-12-08:
What vetsion sdrplay api do you have installed? Should be 3.15

Try attaching your log file.

jcfpi, 2024-12-09:
Windows RSP API 3.15. Where is log file ?
thank you :)

jcfpi, 2024-12-11:
[sdrangel.log](https://github.com/user-attachments/files/18098300/sdrangel.log)

jcfpi, 2024-12-12:
Thank you for help
[image]

srcejon, 2024-12-12:
What's in Workspace 0? (i.e. click W0 on the left)

srcejon, 2024-12-12:
So you need to select SDRPlayV3[0].

jcfpi, 2024-12-12:
W0 is empty.
OK with SDRPlayV3 Thank you.
I wanted to find RSPdx like in other software

closed 2025-01-25 by srcejon