f4exb/sdrangel #2359: Auto run and play when ubuntu boot.

Hsu-Chia-Wei, 2024-12-12:
Hi , 
I would like to know does the SDRangel cli have a auto run and play function on ubuntu 20.04 LTS.

Cause My device run as an IOT device. 

So I would like to deploy after first setting.

Thanks a lot

f4exb, 2024-12-18:
> Cause My device run as an IOT device

If you don't use the GUI I suggest to use the server flavor sdrangelsrv. Then you can use ther REST API to set it up in a Python or any other language script.

Hsu-Chia-Wei, 2024-12-19:
Good to hear that. It will fit my need. Does it have start guide for the Server Flavor sdrangelsvr. 
far now I just use sdragnel GUI and didn't use other version yet

f4exb, 2024-12-26:
You have examples in Python on how the REST API can be used in the scriptsapi folder
There is also a n interactive Swagger documentation at http://<your IP address and port>/doc/swagger-ui/index.html

closed 2025-02-10 by github-actions[bot]