f4exb/sdrangel #2360: Win 11

ThermoNuclear65, 2024-12-14:
SDRAngel stopped working after Win 11 update , splash screen appears then just stops.No messages.

zzzztj, 2024-12-15:
The same on Windows 10 after the update.  I tried deleting Registry, Deleting %appdata%, uninstalling and reinstalling latest version.

Faulting application name: sdrangel.exe, version: 0.0.0.0, time stamp: 0x67516127
Faulting module name: MSVCP140.dll, version: 14.29.30133.0, time stamp: 0x60ff28cc
Exception code: 0xc0000005
Fault offset: 0x0000000000013020
Faulting process id: 0x26e8
Faulting application start time: 0x01db4ea2bf66f4b6
Faulting application path: C:\Program Files\SDRangel\sdrangel.exe
Faulting module path: C:\WINDOWS\SYSTEM32\MSVCP140.dll
Report Id: c0c04a04-019d-49f3-93fa-cf10f8959ac0
Faulting package full name: 
Faulting package-relative application ID: 

Version=1
EventType=APPCRASH
EventTime=133787074924998965
ReportType=2
Consent=1
UploadTime=133787074930905174
ReportStatus=268435456
ReportIdentifier=9af8219e-509b-4c9f-ad92-0b039fed2d08
IntegratorReportIdentifier=c0c04a04-019d-49f3-93fa-cf10f8959ac0
Wow64Host=34404
NsAppName=sdrangel.exe
AppSessionGuid=000026e8-0001-00f4-b6f4-66bfa24edb01
TargetAppId=W:0006a352f8592617c3b8ba5892583f472fad0000ffff!00006b1985e73f8eb68bdd141b1361a7fa9454e14a3a!sdrangel.exe
TargetAppVer=2024//12//05:08:15:35!0!sdrangel.exe
BootId=4294967295
TargetAsId=601
UserImpactVector=808452368
IsFatal=1
EtwNonCollectReason=1
Response.BucketId=60fd1975414ff816344109af51afa36e
Response.BucketTable=4
Response.LegacyBucketId=1459458402839143278
Response.type=4
Sig[0].Name=Application Name
Sig[0].Value=sdrangel.exe
Sig[1].Name=Application Version
Sig[1].Value=0.0.0.0
Sig[2].Name=Application Timestamp
Sig[2].Value=67516127
Sig[3].Name=Fault Module Name
Sig[3].Value=MSVCP140.dll
Sig[4].Name=Fault Module Version
Sig[4].Value=14.29.30133.0
Sig[5].Name=Fault Module Timestamp
Sig[5].Value=60ff28cc
Sig[6].Name=Exception Code
Sig[6].Value=c0000005
Sig[7].Name=Exception Offset
Sig[7].Value=0000000000013020
DynamicSig[1].Name=OS Version
DynamicSig[1].Value=10.0.19045.2.0.0.256.48
DynamicSig[2].Name=Locale ID
DynamicSig[2].Value=2057
DynamicSig[22].Name=Additional Information 1
DynamicSig[22].Value=2e43
DynamicSig[23].Name=Additional Information 2
DynamicSig[23].Value=2e433c85f4e57ef67484a0f0d493005b
DynamicSig[24].Name=Additional Information 3
DynamicSig[24].Value=8224
DynamicSig[25].Name=Additional Information 4
DynamicSig[25].Value=82243062250e3969ec710f2a7249e40b
UI[2]=C:\Program Files\SDRangel\sdrangel.exe
LoadedModule[0]=C:\Program Files\SDRangel\sdrangel.exe
LoadedModule[1]=C:\WINDOWS\SYSTEM32\ntdll.dll
LoadedModule[2]=C:\WINDOWS\System32\KERNEL32.DLL
LoadedModule[3]=C:\WINDOWS\System32\KERNELBASE.dll
[... 387 lines cut ...]
OsInfo[28].Key=wpbuildmin
OsInfo[28].Value=0
OsInfo[29].Key=osver
OsInfo[29].Value=10.0.19041.5247.amd64fre.vb_release.191206-1406
OsInfo[30].Key=buildflightid
OsInfo[30].Value=FF95561B-934D-4F58-AF55-CABBDFFB1A92.1
OsInfo[31].Key=edition
OsInfo[31].Value=Professional
OsInfo[32].Key=ring
OsInfo[32].Value=Retail
OsInfo[33].Key=expid
OsInfo[33].Value=CENSORED ASK if required
OsInfo[34].Key=fconid
OsInfo[35].Key=containerid
OsInfo[36].Key=containertype
OsInfo[37].Key=edu
OsInfo[37].Value=0
FriendlyEventName=Stopped working
ConsentKey=APPCRASH
AppName=sdrangel.exe
AppPath=C:\Program Files\SDRangel\sdrangel.exe
NsPartner=windows
NsGroup=windows8
ApplicationIdentity=2A83FC8C5A576F29AC7E82A219F0138F
MetadataHash=1437616453

srcejon, 2024-12-15:
The last time this happened, installing the latest VC Redistributable helped:

https://aka.ms/vs/17/release/vc_redist.x64.exe

zzzztj, 2024-12-15:
I installed another program that installed the VC as part of the dist, and magically, it started working again.  Thanks for the reply.

ThermoNuclear65, 2024-12-15:
@zzzztj  @srcejon , thanks guys,  the reinstall of the VC_Redist solved the problem , thanks very much !

srcejon, 2024-12-24:
PR to update redistributables to latest version: https://github.com/f4exb/sdrangel-windows-libraries/pull/28

f4exb, 2025-02-02:
Deployed in v7.22.6

closed 2025-02-02 by f4exb