f4exb/sdrangel #2364: SDRangel targets libuhd 4.6.0 while libuhd 4.7.0 is available

Zduniusz, 2024-12-21:
Hello, from Jun 19 [UHD 4.7.0.0](https://github.com/EttusResearch/uhd/releases/tag/v4.7.0.0) is available and many Linux repositories already have it. Since SDRangel expects libuhd.so=4.6.0 it breaks compatibility with many other applications like URH which expect UHD to be 4.7.0.  

[Arch Repo
](https://archlinux.org/packages/extra/x86_64/libuhd/)

```
$ sdrangel --version
SDRangel 7.22.5
```

srcejon, 2024-12-24:
PR for Windows: https://github.com/f4exb/sdrangel-windows-libraries/pull/27

srcejon, 2024-12-24:
> Since SDRangel expects libuhd.so=4.6.0 

What binary release are you using?

srcejon, 2025-01-25:
Have updated the wiki and external/CMakeLists.txt uses 4.7.0 as the tag. Linux github actions use libuhd-dev with no version, as does debian/control and .appveyor.

Just need to updated the snap.

f4exb, 2025-02-02:
Deployed in v7.22.6

closed 2025-02-02 by f4exb