f4exb/sdrangel #2366: rigctl basic support

sq6emm, 2024-12-24:
Hey I am trying to use rigctl support... but its seems its support is experimental...
Is there any chance for improvements?

Lemme show the example below that should be self explanatory:

`

$ rigctl -vvvvvvvvvv -m 2 -r 127.0.0.1:4532
../../tests/rigctl.c(445) Startup: rigctl -vvvvvvvvvv -m 2 -r 127.0.0.1:4532
rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0x784fff48c260, p2=0x784fff4927e0, rig_model=0x784fff48c260, macro_name=0x784fff4927e0
initrigs4_dummy: _init called
rig_init: rig_model=Hamlib NET rigctl
rig_init: rx_range_list1 is empty, using rx_range_list2
rig_init: rig does not have tx_range!!
netrigctl_init version 20230328.0
 1:rig.c(817):rig_open entered
rig_settings_get_path: path=/home/dawszy/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/dawszy/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/dawszy
rig_open: /home/dawszy/hamlib_settings does not exist
rig_open: async_data_enable=1, async_data_supported=0
rig_open: using network address 127.0.0.1:4532
network_open: hoststr=127.0.0.1, portstr=4532
network_open: client port=52420
  2:rig.c(7481):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
  2:rig.c(7488):async_data_handler_start returning(0) 
rig.c(254):add_opened_rig returning2(0) 
rig_open: 0x5ad2b04b544c rs->comm_state==1?=1
  2:rig.c(6084):rig_get_powerstat entered
../../src/rig.c(6104) trace
netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 8 characters, direct=1
0000    52 50 52 54 20 2d 34 0a                             RPRT -4.        
netrigctl_get_powerstat: PS command failed (ret=-4) so returning RIG_POWER_ON
  2:rig.c(6109):rig_get_powerstat returning(0) 
  2:netrigctl.c(261):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
network_flush: network data clear d: ret=0, len=8, ''
network_flush: network data cleared: ret=0, len_read=8/0x8
0000    52 50 52 54 20 2d 34 0a                             RPRT -4.        
write_block(): TX 9 bytes, method=2
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 8 characters, direct=1
0000    52 50 52 54 20 2d 34 0a                             RPRT -4.        
netrigctl_open: chk_vfo error: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 8 characters, direct=1
0000    52 50 52 54 20 2d 34 0a                             RPRT -4.        
netrigctl_get_powerstat: PS command failed (ret=-4) so returning RIG_POWER_ON
  2:rig.c(6109):rig_get_powerstat returning(0) 
  2:netrigctl.c(261):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
network_flush: network data clear d: ret=0, len=8, ''
network_flush: network data cleared: ret=0, len_read=8/0x8
0000    52 50 52 54 20 2d 34 0a                             RPRT -4.        
write_block(): TX 9 bytes, method=2
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 8 characters, direct=1
0000    52 50 52 54 20 2d 34 0a                             RPRT -4.        
Feature not implemented
Feature not implemented

netrigctl_open: vfo_mode=0
netrigctl_transaction: called len=12
rig_flush: called for network device
network_flush called
network_flush: network data clear d: ret=0, len=8, ''
network_flush: network data cleared: ret=0, len_read=8/0x8
0000    52 50 52 54 20 2d 34 0a                             RPRT -4.        
write_block(): TX 12 bytes, method=2
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.    
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 8 characters, direct=1
0000    52 50 52 54 20 2d 34 0a                             RPRT -4.        
  2:netrigctl.c(299):netrigctl_open returning(-4) Feature not implemented

  2:rig.c(7525):async_data_handler_stop entered
  2:rig.c(7555):async_data_handler_stop returning(0) 
network_close: close socket ret=0
 1:rig.c(1301):rig_open returning(-4) Feature not implemented

rig_open: error = 127.0.0.1:4532 No such file or directory

`

and just to make sure it works... but is not usable from any software that supports hamlib

`

$ telnet 0 4532
Trying 0.0.0.0...
Connected to 0.
Escape character is '^]'.
f
435000000
^]
telnet> quit
Connection closed.

`

unfgr, 2025-01-23:
It looks like the rigctl plugin no longer supports newer Hamlib Versions. I'm getting the exact same message on wsjtx when i try to connect to SDRAngel rigctl Server

[image]

sq6emm, 2025-03-04:
This is still valid issue...

closed 2025-04-20 by github-actions[bot]