f4exb/sdrangel #2368: Android IQ input is not stereo

pmate955, 2024-12-27:
I have an Axagon ADA-17 soundcard, and I'm trying to get it to work with a Belka receiver (40KHz IQ output). The soundcard works great on PC, but on Android, it looks like the application can't use it as stereo input, only mono. This causes a lot of mirror signals and it doesn't work as expected. I'm using a Xiaomi Redmi Note 13 Pro 5G (Android 14). 
I don't know if this is the limitation of the Android itself, or the Xiaomi did something, or anything else.
[image]

Thank you! HA8MZ

pmate955, 2024-12-27:
Update: I downloaded an app, called Stereo Sound Recorder, and I recorded raw audio. Then I selected the wav file as FileInput in SdrAngel, and it worked fine, so this means, the device works correctly and it can record stereo input only the SdrAngel can't use it properly.

srcejon, 2024-12-27:
What do you have for the channel map on the device settings - Mono or I=L, Q=R?

pmate955, 2024-12-27:
I pushed through everything I could in the AudioInput settings panel. 
[image]

srcejon, 2024-12-27:
Are there any other Devices listed? Would guess a mic isn't stereo.

pmate955, 2024-12-27:
There isn't. Mic, camcorder, voicerecognition, voicecommunication. There is no change if I plug in the sound card. But only the mic input works, the others are causing crash in the application.

srcejon, 2024-12-27:
> There is no change if I plug in the sound card. 

You'd need to press the refresh button, but probably better to have it connected before SDRangel starts.
What's the device called in the other app?

pmate955, 2024-12-27:
I tried the refresh and restart, but nothing new. 
The other app doesn't need any extra settings. It looks like using as the default audio source.

pmate955, 2024-12-27:
[image]

pmate955, 2024-12-29:
I found this. The other application uses this "unprocessed" input as the source, which works great. But I didn't find anything similar in QAudioFormat. I'm unfamiliar with Qt and C++, so I'm not sure if it's possible to use this instead of the native API.
[image]

closed 2025-02-13 by github-actions[bot]