f4exb/sdrangel #2372: URL for GOES satellite file is broken as has changed in server

NeilJared, 2025-01-10:
When I try to get satellite data, SDRangel says GOES.txt file is missing, and I have checked the url for NORAD files has changed since last september; now GOES, as tle or table, is available here: https://celestrak.org/NORAD/elements/gp.php?GROUP=goes&FORMAT=tle

I cannot change on my SDRangel v.7.22 windows version.

IngmarG, 2025-01-29:
Same problem.

The new URL for the data feed is known, but when I change it in the settings, it always reverts back to the default when the program is restarted.
So where is the default stored?
I have searched just about any possible place, even registry, but couldn't find it.

srcejon, 2025-01-29:
How are you opening the satellite tracker?

By Add Feature on the tool bar or something else?

IngmarG, 2025-01-30:
> How are you opening the satellite tracker?
> 
> By Add Feature on the tool bar or something else?

By using the preset configuration

[image]

IngmarG, 2025-02-16:
Any new comment?

IngmarG, 2025-03-11:
Nothing?
Shouldn't be that difficult if only we knew how to save the settings properly.

IngmarG, 2025-04-21:
Bumping...
Still no solution or suggestion?

srcejon, 2025-04-22:
> And to change the settings, I use the TLE tab in the triple bar menu of the satellite tracker. But those changes are not persistent and revert back to the preset.

Did you save the preset? Press the Star Button in the toolbar then select the preset and press the Update button (second from left).

IngmarG, 2025-04-22:
Thx, that did it.

Not really obvious.
Without your reply, I would never have found out.

But, problem solved., 😉 .

 

Kind regards,

 

Ingmar

IngmarG, 2025-04-22:
> > And to change the settings, I use the TLE tab in the triple bar menu of the satellite tracker. But those changes are not persistent and revert back to the preset.
> 
> Did you save the preset? Press the Star Button in the toolbar then select the preset and press the Update button (second from left).

Thx, that did it.

Not really obvious.
Without your reply, I would never have found out.
But, problem solved., 😉 .