f4exb/sdrangel #2375: SSB Modulator - cannot send voice - version 7.22.1, cannot send CW or Mic Modulation in later releases

testpuddle, 2025-01-16:
I downloaded and tested all Windows versions from 7.22.0 and later to check if I could find which relkease broke the SSB Modulator.

7.22.0 CW send and Mic modulation works as expected when tested on Windows 11
7.22.1 CW Send works, Mic modulation does not work when tested on Windows 11
7.22.2 CW send and Mic modulation do not work when tested on Windows 11
7.22.3 CW send and Mic modulation do not work when tested on Windows 11
7.22.4 CW send and Mic modulation do not work when tested on Windows 11
7.22.5 CW send and Mic modulation do not work when tested on Windows 11

Interim workaround - run 7.22.0

Beautiful program BTW.

testpuddle, 2025-01-18:
ADALM-Pluto running under Microsoft Windows 11 Home Version 10.0.22631 Build 22631.

Thanks.

makrohard, 2025-01-19:
I can confirm that behaviour on a Fedora Linux System, using HackRF One as SDR. The distributor-shipped Version 7.22.5 does not work as expected, neither does this version, when compiled from source.

The CW Keyer - from text or Mouse Keyboard keyer do not work, neither does the audio input. Tone modulation or file input do work fine, though.

Using 7.22.0, compiled from source, everything works as expected.

srcejon, 2025-01-20:
Mic/audio input works for me on Windows with 7.22.5 with a USRP. It was not working in some earlier 7.22.x versions. Can you clarify what you mean by "do not work"? No modulation on spectrum? No audio feedback? A screenshot would help.

The Morse keyer is not working for me in the SSB demod (nothing in channel/device spectrum - no audio). It works OK in AM and NFM mods.

srcejon, 2025-01-20:
The above patch should fix the Morse keyer in the SSB Mod. 

I don't have a Pluto nor HackRF to investigate those problems, I'm afraid.

makrohard, 2025-01-20:
I applied the above fix #2380 (one line) to the latest version from github and compiled sdrangel. Morse keyer - as well as microphone input - do now both work as expected, using HackRF as SDR on Fedora 41. 

Description of observed symptoms in the unpatched version (distributor-shipped 7.22.5 on Fedora 41, as well as latest source from github, without #2380 ): 

* Starting morse-keyer or audio-input does start the waterfall on the spectrum channel, but without any signal. To observe a 'running' waterfall, I first had to put a short tone-signal, that starts scrolling down on activating keyer or mic-input. No-signal alone has not been observable.
* After starting morse-keyer, none of the buttons (choosing text, keyboard or mouse) did generate a signal or have any observable effect. No debug-output on loglevel Info.
* After starting mic-input, the waterfall in the channel-spectrum starts so run, like above, without signal. I double checked the mike. Debug-output: '(I) AudioInputDevice::start: audio format OK'

srcejon, 2025-01-20:
> I applied the above fix [#2380](https://github.com/f4exb/sdrangel/pull/2380) (one line) to the latest version from github and compiled sdrangel. Morse keyer - as well as microphone input - do now both work as expected, using HackRF as SDR on Fedora 41.

I don't think this patch should have changed anything to do with the microphone input, only the Morse Keyer. (But as I said, Mic works ok for me in 7.22.5).

HackRF problems are probably a separate issue: https://github.com/f4exb/sdrangel/issues/2283

makrohard, 2025-01-20:
I agree, that the #2380 fix is not very likely to fix anything but the keyer. So made a fresh clone of https://github.com/f4exb/sdrangel.git (version number showing in app is 7.22.5-g83b36c6aa, QT5, 64 bit). After compilinng that version the mic-input problem could not be observed anymore: Mic-input does work as expected (keyer does not). I can still observe 'mic-input not working' on the fedora41-shipped 7.22.5, though.

After applying the patch and re-compiling, the keyer does work as expected. As well does mic-input.

on the demodulater, i recieve only noise in both versions, that might be #2283 related

srcejon, 2025-01-20:
I forgot - there's a patch on master that isn't in 7.22.5 that fixes the SSB mod when channel sample rate is higher than audio sample rate. I was using master earlier, not 7.22.5 That's likely the problem here - so both SSB issues should be fixed in the next release.

srcejon, 2025-02-07:
Closing as reported fixed: https://github.com/f4exb/sdrangel/pull/2380#issuecomment-2641363141