f4exb/sdrangel #2376: SDRPlay Clone 0x1DF7/0x2500 is not recognized on Mac

saschaludwig, 2025-01-16:
When I try to use the Mirics chipset based blue (dolphin) MSI.SDR receiver, it isn't listed in the device dialogue on Mac x64 in SDRangel Version 7.22.4:

[image]

The USB Prober on Mac however recognizes the USB device:
[image]

Any Idea what I can do to find the issue?

This is the device in question:
[image]

saschaludwig, 2025-01-16:
I think I just found the info, that the SDRplay isn't supported (anymore) in the MacOS build?
https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/sdrplay/readme.md

srcejon, 2025-01-17:
The Mac build uses the SDRplay v3 driver, that uses SDRplay's API, so clones probably not supported unless you build it yourself.

Axpelle, 2025-01-26:
Have a look at this warning from SDRPlay: https://www.sdrplay.com/fakealert/

saschaludwig, 2025-01-26:
> Have a look at this warning from SDRPlay: https://www.sdrplay.com/fakealert/

How does that fix the problem?
I also have a SDRplay RSPdx that works totally fine. But that's not the point. My issue was, that I wanted to use the Mirics SDR device with sdrangel.

Axpelle, 2025-01-27:
There are numerous references regarding that SDR to the effect that it uses fake or cloned Mirics chipset. In particular Mirics only sells to OEMs and ODMs and I very much doubt that this SDR is one of them. There may be workarounds or more or less successful attempts at reverse-engineering the drivers. But more likely than not the level of performance will fall far short of that obtainable from the genuine chipset. So yes that won't fix your problem.

closed 2025-01-27 by saschaludwig