f4exb/sdrangel #2378: SondeHub listener endpoint high request rate

TheSkorm, 2025-01-19:
[image]

I think there might be a bug or lack of min reporting time in the listeners position updates in sdrangel. We are seeing signficantly more load from sdrangel from 2 callsigns against our service - both using sdrangel. Each station is updating its location 200 times a minute.

[image]

srcejon, 2025-01-19:
Sorry about that - what min time would you like it to use?

TheSkorm, 2025-01-19:
For mobile, 30 seconds. For fixed 5 minutes. Thank you!

PR #2379 (2025-01-20): Radiosonde: Rate limit position updates to SondeHub.
Fixes #2378
commit a3692309 (2025-01-20): Radiosonde: Rate limit position updates to SondeHub. Fixes #2378.
closed 2025-01-20 by f4exb