f4exb/sdrangel #2391: Frequency scanner causing crash

RyanHope, 2025-01-29:
I tried following the video on youtube demonstrating importing channel markers for use in the frequency scanner. Holy sh*t this is a pain in the butt. I've got ~25 channels in the 70cm band range. Load the spectrum markers, go to frequency scanner, add range, set 440mhz to 465mhz. Now I have a 2000 entries in the scanner list with lots of "duplicate" annotations where there will be the correct entry and then the adjacent frequency also with that same annotation. So, I try and manually prune list list down, select a bunch of empty channels I don't care about, click remote, sometimes it works but it eventually crashes and I have to start this all over.

[image]

So in this image 444.075mhz was the imported annotation.  There has to be an easier way to do this.

Axpelle, 2025-01-29:
Why don't you just enter the 25 channels manually? Use the "Add" button instead of "Add Range"

RyanHope, 2025-01-29:
Who wants to enter channels manually when most people already have them in a spreadsheet for use with other radios? 25 channels is nothing, there are hundreds more I might want to import.

Axpelle, 2025-01-29:
I’ve entered all VHF marine frequencies with “Add Range” and haven’t 
experienced the issues you mention. I’m on Mac Mini M2 chip using latest 
version of SDRA

———— Original message received from "Ryan Hope" 
***@***.***> on 29/01/2025 07:00:24 ————

Subject Re: [f4exb/sdrangel] Frequency scanner causing crash (Issue 
#2391)

RyanHope, 2025-01-29:
I am on a AMD Ryzen 9 7900X w/ 64gb of ram running win10.

srcejon, 2025-01-29:
If you have hundreds of frequencies in a spreadsheet, the best way to import them would be via the API.

Create a scanner.json file with them listed like this:

        {
            "FreqScannerSettings": {
                "frequencies": [
                    {
                        "enabled": 1,
                        "frequency": 250000000
                    },
                    {
                        "enabled": 1,
                        "frequency": 251000000
                    }
                ]
            },
            "channelType": "FreqScanner",
            "direction": 0
        }
        
Then run:

    curl -X PATCH "http://127.0.0.1:8091/sdrangel/deviceset/0/channel/1/settings" -d @scanner.json

Replacing deviceset/0/channel/1 with the Frequency Scanners Device Set and Channel Index.

srcejon, 2025-01-29:
As to the crash, enter your channels then save the settings before pressing run and restart SDRangel. That way, even if it crashes, you wont need to enter them again.

In order to debug the crash, you'll need to provide more info such as full screenshots of when it crashed and debug log (Preferences > logging).

RyanHope, 2025-01-29:
So you want a screenshot of an empty desktop? Cause that's what I get when
the app crashes. No other feedback.

closed 2025-01-29 by srcejon

RyanHope, 2025-01-29:
What a joke. Just gonna ignore the crash?

neurogenesis, 2025-06-21:
I ran into a similar issue when trying to prune a list of Marine frequencies added via range. I realized I had intervals at 25k Hz instead of 50k Hz, but I had already added notes to indicate the channels, so I thought it'd be faster to just remove the in-between channels than re-create and re-annotate.

If I select a cell within the row, I'm able to click "Remove" without any problems.

However, if I select the entire row and click "Remove", then the app reliably crashes.

I'm on macOS 15.5.0 (Apple M3).

@RyanHope : In an effort to be more considerate and helpful to the team volunteering their time to this massive project, can you confirm how you deleted those individual rows and verify if you're seeing the same behavior?

RyanHope, 2025-06-22:
@neurogenesis No I am done with this trash project

srcejon, 2025-06-23:
> However, if I select the entire row and click "Remove", then the app reliably crashes.

Ok, that's a bug thanks. Should be easy to fix.