f4exb/sdrangel #2392: MAcOS: Stcuk at DeviceEnumerator::enumerateDevices: sdrangel.samplesource.airspy

ArtemKiyashko, 2025-01-29:
Restart SDRangel after using RTL-SDR pretty much often causing a stuck at 'loading plugins' step. When running in terminal, then this log in output:
```
./SDRangel --scratch
2025-01-29 14:37:28.118 (I) SDRangel 7.22.5 Qt 6.7.3 64b arm64 macOS Sonoma (14.6) DSP Rx:24b Tx:16b PID 47962
2025-01-29 14:37:28.119 (I) MainSettings::MainSettings: settings file: format: 0 location: /Users/artem/Library/Preferences/com.f4exb.SDRangel.plist
2025-01-29 14:37:28.119 (D) MessagePipes::startGC
2025-01-29 14:37:28.119 (D) DataPipes::startGC
2025-01-29 14:37:28.119 (D) MainCore::initPosition: Using position source "corelocation"
2025-01-29 14:37:28.119 (W) Could not find permission plugin for QLocationPermission. Please make sure you have included the required usage description in your Info.plist
2025-01-29 14:37:28.119 (W) MainCore::positionError:  QGeoPositionInfoSource::AccessError
2025-01-29 14:37:28.119 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2025-01-29 14:37:28.175 (D) AudioDeviceManager::AudioDeviceManager: input device #0: MacBook Pro Microphone
2025-01-29 14:37:28.175 (D) AudioDeviceManager::AudioDeviceManager: input device #1: iPhone Microphone
2025-01-29 14:37:28.176 (D) AudioDeviceManager::AudioDeviceManager: input device #2: AUKEY PC—W1
2025-01-29 14:37:28.176 (D) AudioDeviceManager::AudioDeviceManager: input device #3: Microsoft Teams Audio
2025-01-29 14:37:28.176 (D) AudioDeviceManager::AudioDeviceManager: input device #4: ZoomAudioDevice
2025-01-29 14:37:28.176 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2025-01-29 14:37:28.179 (D) AudioDeviceManager::AudioDeviceManager: output device #0: MacBook Pro Speakers
2025-01-29 14:37:28.179 (D) AudioDeviceManager::AudioDeviceManager: output device #1: DELL U2422H
2025-01-29 14:37:28.179 (D) AudioDeviceManager::AudioDeviceManager: output device #2: DELL U2421HE
2025-01-29 14:37:28.179 (D) AudioDeviceManager::AudioDeviceManager: output device #3: Microsoft Teams Audio
2025-01-29 14:37:28.179 (D) AudioDeviceManager::AudioDeviceManager: output device #4: ZoomAudioDevice
2025-01-29 14:37:28.179 (D) MainWindow::MainWindow: start
2025-01-29 14:37:29.313 (D) MainWindow::MainWindow: load settings...
2025-01-29 14:37:29.313 (D) MainWindow::loadSettings
2025-01-29 14:37:29.313 (D) AudioDeviceManager::deserialize
2025-01-29 14:37:29.313 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "MacBook Pro Speakers"  sampleRate:  48000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  false  udpChannelMode:  0  udpChannelCodec:  0  decimationFactor:  1
2025-01-29 14:37:29.313 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "System default device"  sampleRate:  48000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  false  udpChannelMode:  0  udpChannelCodec:  0  decimationFactor:  1
2025-01-29 14:37:29.314 (D) FFTEngine::create: using FFTW engine
2025-01-29 14:37:29.314 (D) FFTWEngine::configure: no FFTW wisdom file
2025-01-29 14:37:29.316 (D) FFT: creating FFTW plan (n=128,forward) took 2 ms
2025-01-29 14:37:29.316 (D) FFTEngine::create: using FFTW engine
2025-01-29 14:37:29.316 (D) FFTWEngine::configure: no FFTW wisdom file
2025-01-29 14:37:29.320 (D) FFT: creating FFTW plan (n=256,forward) took 4 ms
2025-01-29 14:37:29.320 (D) FFTEngine::create: using FFTW engine
2025-01-29 14:37:29.320 (D) FFTWEngine::configure: no FFTW wisdom file
2025-01-29 14:37:29.329 (D) FFT: creating FFTW plan (n=512,forward) took 8 ms
2025-01-29 14:37:29.329 (D) FFTEngine::create: using FFTW engine
2025-01-29 14:37:29.329 (D) FFTWEngine::configure: no FFTW wisdom file
2025-01-29 14:37:29.343 (D) FFT: creating FFTW plan (n=1024,forward) took 13 ms
2025-01-29 14:37:29.343 (D) MainWindow::MainWindow: load plugins...
2025-01-29 14:37:29.343 (D) PluginManager::loadPluginsPart folder /Applications/SDRangel.app/Contents/MacOS/../lib/sdrangel/plugins is empty
2025-01-29 14:37:29.343 (D) PluginManager::loadPluginsPart folder /Applications/SDRangel.app/Contents/MacOS/lib/plugins is empty
2025-01-29 14:37:29.345 (D) PluginManager::loadPluginsDir: fileName: libantennatools.dylib
2025-01-29 14:37:29.346 (I) PluginManager::loadPluginsDir: loaded plugin libantennatools.dylib
2025-01-29 14:37:29.346 (D) PluginManager::loadPluginsDir: fileName: libaprs.dylib
2025-01-29 14:37:29.347 (I) PluginManager::loadPluginsDir: loaded plugin libaprs.dylib
2025-01-29 14:37:29.347 (D) PluginManager::loadPluginsDir: fileName: libbeamsteeringcwmod.dylib
2025-01-29 14:37:29.348 (I) PluginManager::loadPluginsDir: loaded plugin libbeamsteeringcwmod.dylib
2025-01-29 14:37:29.348 (D) PluginManager::loadPluginsDir: fileName: libchanalyzer.dylib
2025-01-29 14:37:29.349 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.dylib
2025-01-29 14:37:29.349 (D) PluginManager::loadPluginsDir: fileName: libchannelpower.dylib
2025-01-29 14:37:29.349 (I) PluginManager::loadPluginsDir: loaded plugin libchannelpower.dylib
2025-01-29 14:37:29.349 (D) PluginManager::loadPluginsDir: fileName: libdemodadsb.dylib
2025-01-29 14:37:29.353 (I) PluginManager::loadPluginsDir: loaded plugin libdemodadsb.dylib
2025-01-29 14:37:29.353 (D) PluginManager::loadPluginsDir: fileName: libdemodais.dylib
2025-01-29 14:37:29.353 (I) PluginManager::loadPluginsDir: loaded plugin libdemodais.dylib
2025-01-29 14:37:29.353 (D) PluginManager::loadPluginsDir: fileName: libdemodam.dylib
2025-01-29 14:37:29.354 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.dylib
2025-01-29 14:37:29.354 (D) PluginManager::loadPluginsDir: fileName: libdemodapt.dylib
2025-01-29 14:37:29.355 (I) PluginManager::loadPluginsDir: loaded plugin libdemodapt.dylib
2025-01-29 14:37:29.355 (D) PluginManager::loadPluginsDir: fileName: libdemodatv.dylib
2025-01-29 14:37:29.356 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.dylib
2025-01-29 14:37:29.356 (D) PluginManager::loadPluginsDir: fileName: libdemodbfm.dylib
2025-01-29 14:37:29.356 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.dylib
2025-01-29 14:37:29.356 (D) PluginManager::loadPluginsDir: fileName: libdemodchirpchat.dylib
2025-01-29 14:37:29.357 (I) PluginManager::loadPluginsDir: loaded plugin libdemodchirpchat.dylib
2025-01-29 14:37:29.357 (D) PluginManager::loadPluginsDir: fileName: libdemoddab.dylib
2025-01-29 14:37:29.358 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddab.dylib
2025-01-29 14:37:29.358 (D) PluginManager::loadPluginsDir: fileName: libdemoddatv.dylib
[... 312 lines cut ...]
2025-01-29 14:37:29.603 (D) PluginManager::registerFeature  RigCtl Server  with channel name  "sdrangel.feature.rigctlserver"
2025-01-29 14:37:29.603 (D) PluginManager::registerFeature  Rotator Controller  with channel name  "sdrangel.feature.gs232controller"
2025-01-29 14:37:29.603 (D) PluginManager::registerFeature  SID  with channel name  "sdrangel.feature.sid"
2025-01-29 14:37:29.603 (D) PluginManager::registerRxChannel  SSB Demodulator  with channel name  "sdrangel.channel.ssbdemod"
2025-01-29 14:37:29.603 (D) PluginManager::registerTxChannel  SSB Modulator  with channel name  "sdrangel.channeltx.modssb"
2025-01-29 14:37:29.603 (D) PluginManager::registerFeature  Satellite Tracker  with channel name  "sdrangel.feature.satellitetracker"
2025-01-29 14:37:29.603 (D) PluginManager::registerFeature  Simple PTT  with channel name  "sdrangel.feature.simpleptt"
2025-01-29 14:37:29.603 (D) PluginManager::registerFeature  Sky Map  with channel name  "sdrangel.feature.skymap"
2025-01-29 14:37:29.603 (D) PluginManager::registerFeature  Star Tracker  with channel name  "sdrangel.feature.startracker"
2025-01-29 14:37:29.603 (D) PluginManager::registerSampleMIMO  Test Multiple Input  with MIMO name  sdrangel.samplemimo.testmi  and hardware id  "TestMI"
2025-01-29 14:37:29.603 (D) PluginManager::registerSampleSink  Test Sink Output  with sink name  sdrangel.samplesink.testsink  and hardware id  "TestSink"
2025-01-29 14:37:29.603 (D) PluginManager::registerSampleSource  Test Source input  with source name  sdrangel.samplesource.testsource  and hardware id  "TestSource"
2025-01-29 14:37:29.603 (D) PluginManager::registerSampleMIMO  Test Synchronous Multiple Output  with MIMO name  sdrangel.samplemimo.testmosync  and hardware id  "TestMOSync"
2025-01-29 14:37:29.603 (D) PluginManager::registerRxChannel  UDP Channel Sink  with channel name  "sdrangel.channel.udpsink"
[... cut ...]
2025-01-29 14:37:29.604 (D) PluginManager::registerRxChannel  WDSP Receiver  with channel name  "sdrangel.channel.wdsprx"
2025-01-29 14:37:29.604 (D) PluginManager::registerRxChannel  WFM Demodulator  with channel name  "sdrangel.channel.wfmdemod"
2025-01-29 14:37:29.604 (D) PluginManager::registerTxChannel  WFM Modulator  with channel name  "sdrangel.channeltx.modwfm"
2025-01-29 14:37:29.604 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.aaroniartsasource
2025-01-29 14:37:29.604 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.airspy
```

srcejon, 2025-01-30:
It should be safe to delete any libinput plugins for devices that you don't want to use. (But of course, you shouldn't need to delete them, this is just to find out where the problem might be).

Would be interesting to see what the log file shows after deleting them.

ArtemKiyashko, 2025-02-14:
Can not reproduce anymore :D But if i get it again i will update this issue

closed 2025-03-29 by ArtemKiyashko