f4exb/sdrangel #2393: Mac OS. Crashes immediately after loading.

Registan29, 2025-01-29:
I just installed the program, launched it, selected the configuration. The program loaded and immediately crashes....

Maybe someone had the same error?

Mac OS Ventura 13.7.2 intel.

srcejon, 2025-01-29:
Which configuration did you select?

What SDR do you have?

What happens if you don't select a config and just add your device manually?

Registan29, 2025-01-29:
The program closes immediately after selecting the startup configuration (the one that needs to be selected when starting the program). I don't even have time to go to the menu or anywhere else. Moreover, when choosing any startup configuration, the result is the same...

I have hackRF one

[image]

I just select any item here, after that the program will crashes

srcejon, 2025-01-29:
What happens if you just press close?

Registan29, 2025-01-29:
Thank you) It works!)

closed 2025-04-27 by github-actions[bot]

srcejon, 2025-06-09:
> im having this problem under NixOS (Linux)

So what happens if you just press Close?

krislyon, 2025-06-15:
I also experienced this on first startup when selecting the APT configuration.  
Happened a few times in a row.  Selecting 'close' on the configuration selection dialog allowed SDRAngel to launch without crashing.  I was then able to open the APT configuration without a crash, but I did get an error that it failed to download GOES data from celestrak.  Might be an issue with the error handling on startup?