f4exb/sdrangel #2396: Remoteoutput for Windows?

gvanem, 2025-02-04:
I'm curious to know why `plugins/samplesink/remoteoutput/UDPSocket.cpp` is not very Windows-friendly
(e.g. includes `<unistd.h>`, `<net/if.h>` etc.).

I see `plugins/samplesink/CMakeLists.txt` has:
```c
if (CM256CC_FOUND AND (HAS_SSE3 OR HAS_NEON))
    add_subdirectory(remoteoutput)
else()
  ...
```

For me, cm256cc builds fine. So should there perhaps be Winsock support for `remoteoutput` too?

I managed to patch the sources to build on Windows:  [diff-1.txt](https://github.com/user-attachments/files/18658502/diff-1.txt)
But have not tested this much.

srcejon, 2025-02-04:
That file isn't included in CMakeLists.txt as far as I can see - so shouldn't be compiled. Can probably be removed.

gvanem, 2025-02-04:
Ah, I see. Some old leftovers...

closed 2025-03-23 by github-actions[bot]