f4exb/sdrangel #2400: Frequent change of CPU% use for no apparent reason

Axpelle, 2025-02-14:
Hello

I've been monitoring CPU usage by SDRA on my Mac mini M2 chip for some time. While most applications will typically hover around a fairly fixed percentage during a given task, I notice that for SDRA it can jump from 55% to over 95% in a matter of seconds and then keep fluctuating widely although there is no significant change in the operation being performed (eg scanning a couple of air band frequencies). Is there a rational explanation to this?

srcejon, 2025-02-14:
A screenshot might help.

Axpelle, 2025-02-14:
> A screenshot might help.

Here are a couple of screenshots taken over a period of about 1 minute while SDRA is scanning four airband frequencies. The CPU% varies from 62 to a peak go 92.3 for no apparent reason.

[image]
[image]
[image]
[image]
[image]

srcejon, 2025-02-14:
I meant one of SDRangel so I can see what you have open and the settings you have :)

Axpelle, 2025-02-14:
> I meant one of SDRangel so I can see what you have open and the settings you have :)

Oops my mistake. Here you are. Note that the ADSB receiver part is deliberately switched off to avoid even higher CPU% usage. Here only the scanner is working.

[image]

srcejon, 2025-02-14:
Is the change when it switches from scanning to just listening to audio?

What happens if the scanning is switched off?

Axpelle, 2025-02-14:
Just checked. With scanning feature off the CPU% still varies from 62 to 89-90. All very random it seems.

Axpelle, 2025-02-14:
And with all receivers switched off it jumps from 32.7 to 58-59 randomly

srcejon, 2025-02-14:
Try closing the ADSB window.

Axpelle, 2025-02-15:
I can also add that even with all devices turned off and SDRA open CPU% can jump from 30 to 60 and back randomly

srcejon, 2025-02-15:
Just to check, you are using the arm64 build on an M2 mac, not x86?

Axpelle, 2025-02-15:
This is correct.

srcejon, 2025-03-01:
I can't reproduce this. For me, when I have SDRangel open with just the RTL SDR device open, but acquisition stopped, Activity Monitor shows CPU usage at around 2.5%.

When opening an AM Demod channel, this jumps to ~32%, even when the device is stopped, so something possibly needs to be looked at there (possibly tick() function in GUI), but it is pretty constant at 32%.

Axpelle, 2025-03-01:
I have noted from trial and error that inactive Workspaces actually cause the CPU usage to rise and fluctuate considerably. After removing 3 inactive Workspaces the lowest CPU figure I can get with all devices and acquisition off is in the range 10-13.5%. With ADSB only on the CPU rises to 42-45%. If I turn the ADSB off and airband scanner on in the same Workspace the CPU is in the range 56-62%. With both ADSB and scanner on in the same Workspace the CPU fluctuates between 68% and 71%. So removing empty Workspaces seems to have eliminated the large CPU fluctuations seen before. However, the overall CPU usage on my side is very far from your 32%. Hope this helps.

closed 2025-04-17 by github-actions[bot]