f4exb/sdrangel #2401: v7.22.6 Windows Build Failing - Need extra libraries

ikjordan, 2025-02-16:
I tried to build the Windows version of SDRAngel in an attempt to help move on #2283

Recently (7.22.2) #2270 changes have been added to use FLAC and ZLib. I've read through the documentation, but cannot find an answer. What additional libraries need to be installed so that the Windows build can complete please? Currently it fails for me as FLAC functions cannot be linked.

I thought I could avoid this by setting `-DENABLE_CHANNELRX_REMOTETCPSINK=OFF`, but the build still tries to link `inputtcpremote.lib`, which also includes references to FLAC.

srcejon, 2025-02-16:
zlib is prebuilt as part of https://github.com/f4exb/sdrangel-windows-libraries which is a submodule in external/windows. Make sure that's up to date.

FLAC is build automatically as an external project in external/CMakeLists.txt

Sounds like your cmakefiles may be a bit out of sync. I'd suggest starting again from a clean build directory then posting the exact error if you get one.

ikjordan, 2025-02-16:
Thanks. I'd built from a clean clone using the command line build steps. The link errors are:
```
[3467/3687] Linking CXX shared library bin\plugins\remotetcpsink.dll
FAILED: bin/plugins/remotetcpsink.dll lib/plugins/remotetcpsink.lib 
cmd.exe /C "cd . && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_dll --intdir=plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100226~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100226~1.0\x64\mt.exe --manifests  -- C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1442~1.344\bin\Hostx64\x64\link.exe /nologo plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsink_autogen\mocs_compilation.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsink.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinkbaseband.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinksink.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinksettings.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinkwebapiadapter.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinkplugin.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinkgui.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinksettingsdialog.cpp.obj  /out:bin\plugins\remotetcpsink.dll /implib:lib\plugins\remotetcpsink.lib /pdb:bin\plugins\remotetcpsink.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO  /INCREMENTAL:NO /LTCG  lib\sdrgui.lib  lib\FLAC.lib  ..\external\windows\zlib\lib\zlibstaticd.lib  lib\sdrbase.lib  C:\Qt\5.15.2\msvc2019_64\lib\Qt5WebSockets.lib  lib\swagger.lib  ..\external\windows\libopus\lib\x64\libopus.lib  ..\external\windows\fftw-3\libfftw3f-3.lib  lib\serialdv.lib  lib\httpserver.lib  lib\qrtplib.lib  ..\external\windows\libsigmf\lib\libsigmf.lib  DbgHelp.lib  ..\external\windows\libsigmf\lib\libsigmf.lib  DbgHelp.lib  C:\Qt\5.15.2\msvc2019_64\lib\Qt5Multimedia.lib  C:\Qt\5.15.2\msvc2019_64\lib\Qt5Positioning.lib  lib\logging.lib  C:\Qt\5.15.2\msvc2019_64\lib\Qt5Network.lib  C:\Qt\5.15.2\msvc2019_64\lib\Qt5OpenGL.lib  C:\Qt\5.15.2\msvc2019_64\lib\Qt5Widgets.lib  C:\Qt\5.15.2\msvc2019_64\lib\Qt5Gui.lib  C:\Qt\5.15.2\msvc2019_64\lib\Qt5Core.lib  opengl32.lib  glu32.lib  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib  && cd ."
LINK: command "C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1442~1.344\bin\Hostx64\x64\link.exe /nologo plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsink_autogen\mocs_compilation.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsink.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinkbaseband.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinksink.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinksettings.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinkwebapiadapter.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinkplugin.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinkgui.cpp.obj plugins\channelrx\remotetcpsink\CMakeFiles\remotetcpsink.dir\remotetcpsinksettingsdialog.cpp.obj /out:bin\plugins\remotetcpsink.dll /implib:lib\plugins\remotetcpsink.lib /pdb:bin\plugins\remotetcpsink.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO /INCREMENTAL:NO /LTCG lib\sdrgui.lib lib\FLAC.lib ..\external\windows\zlib\lib\zlibstaticd.lib lib\sdrbase.lib C:\Qt\5.15.2\msvc2019_64\lib\Qt5WebSockets.lib lib\swagger.lib ..\external\windows\libopus\lib\x64\libopus.lib ..\external\windows\fftw-3\libfftw3f-3.lib lib\serialdv.lib lib\httpserver.lib lib\qrtplib.lib ..\external\windows\libsigmf\lib\libsigmf.lib DbgHelp.lib ..\external\windows\libsigmf\lib\libsigmf.lib DbgHelp.lib C:\Qt\5.15.2\msvc2019_64\lib\Qt5Multimedia.lib C:\Qt\5.15.2\msvc2019_64\lib\Qt5Positioning.lib lib\logging.lib C:\Qt\5.15.2\msvc2019_64\lib\Qt5Network.lib C:\Qt\5.15.2\msvc2019_64\lib\Qt5OpenGL.lib C:\Qt\5.15.2\msvc2019_64\lib\Qt5Widgets.lib C:\Qt\5.15.2\msvc2019_64\lib\Qt5Gui.lib C:\Qt\5.15.2\msvc2019_64\lib\Qt5Core.lib opengl32.lib glu32.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:bin\plugins\remotetcpsink.dll.manifest" failed (exit code 1120) with the following output:
   Creating library lib\plugins\remotetcpsink.lib and object lib\plugins\remotetcpsink.exp
LINK : warning LNK4098: defaultlib 'MSVCRTD' conflicts with use of other libs; use /NODEFAULTLIB:library
remotetcpsinksink.cpp.obj : error LNK2001: unresolved external symbol __imp_FLAC__stream_encoder_new
remotetcpsinksink.cpp.obj : error LNK2001: unresolved external symbol __imp_FLAC__stream_encoder_init_stream
remotetcpsinksink.cpp.obj : error LNK2001: unresolved external symbol __imp_FLAC__stream_encoder_get_state
remotetcpsinksink.cpp.obj : error LNK2001: unresolved external symbol __imp_FLAC__format_blocksize_is_subset
remotetcpsinksink.cpp.obj : error LNK2001: unresolved external symbol __imp_FLAC__stream_encoder_set_compression_level
remotetcpsinksink.cpp.obj : error LNK2001: unresolved external symbol __imp_FLAC__stream_encoder_set_bits_per_sample
[... cut ...]
I've attached the full (4MB) build log as well. To double check I went back to an earlier version (7.22.0), deleted my build directory and that built fine for me from the command line.

I subsequently started again and followed the Visual Studio build Instructions. That build completed correctly. I can run it and reproduce #2283.

So, the current situation is that I cannot build the current version on the command line, but I can build it from within VS. I can build older version on the command line. I am not blocked, as I can continue with builds within VS.

ikjordan, 2025-02-16:
Good spot 👍 
Yes, that fixes it. I've just completed a successful build of the latest code from the command line

PR #2402 (2025-02-16): Fix -DBUILD_SHARED_LIBS=ON for FLAC
May fix #2401
closed 2025-02-16 by f4exb