f4exb/sdrangel #2403: New errors discovered

KanotixPinguin, 2025-02-17:
The map does not save the settings such as deactivating KIWI SDR. After restarting, everything is back on.
The program was tested under Linux and Windows:
I connect to my SDRANGELSERVER Remote TCP via the network or internet. When I close and restart my client program, I always have to restart my server SDRANGELSERVER because no connection is established during play. I hope that this can be fixed quickly. Thanks to the developers for the great program.
If I may point out, the scanner function is very slow.55 73.

srcejon, 2025-02-26:
> The map does not save the settings such as deactivating KIWI SDR. After restarting, everything is back on.

What do you mean by 'restarting'? By exiting SDRangel and starting it again? Or closing the map and opening a new map?

In the former case, the settings should be saved. In the latter case they wont be, as the default settings are loaded each time you open a new map.

To set the default settings for the map (or any other plugin):

 - set the settings you want, 
 - then click the gear icon in the top left of the map window. 
 - Press the star button in the bottom left of the Basic Feature settings dialog
 - Press the plus button
 - Set Group to Defaults
 - Set Description to Default
 - Press OK

srcejon, 2025-02-26:
> I connect to my SDRANGELSERVER Remote TCP via the network or internet. 
> When I close and restart my client program, I always have to restart my 
> server SDRANGELSERVER because no connection is established during play.

What is your client program? What messages do you see in the SDRangel debug log file? (I.e. might be easier to run the server via the GUI to debug what is going on).

srcejon, 2025-02-26:
> If I may point out, the scanner function is very slow

Please give a more detailed description of what settings you are using and how fast you think it should be.

KanotixPinguin, 2025-02-26:
> > Wenn ich darauf hinweisen darf, ist die Scannerfunktion sehr langsam
> 
> Bitte geben Sie eine genauere Beschreibung, welche Einstellungen Sie verwenden und wie schnell Sie denken, dass es sein sollte.

> > If I may point out, the scanner function is very slow
> 
> Please give a more detailed description of what settings you are using and how fast you think it should be.

Hello Thank you for your quick reply. I'll be back in a few days on all posts here regarding the client server logs and also the scanner speed. I can't access it at the moment as I have both laptops in use for data recovery. But I will be available shortly.

KanotixPinguin, 2025-03-01:
> > Die Karte speichert nicht die Einstellungen wie die Deaktivierung von KIWI SDR. Nach dem Neustart ist alles wieder an.
> 
> Was meinen Sie mit dem "Neustart"? Indem Sie SDRangel verlassen und wieder anfangen? Oder die Karte schließen und eine neue Karte öffnen?
> 
> Im ersten Fall sollten die Einstellungen gespeichert werden. Im letzteren Fall werden sie nicht sein, da die Standardeinstellungen jedes Mal geladen werden, wenn Sie eine neue Karte öffnen.
> 
> So legen Sie die Standardeinstellungen für die Karte (oder ein anderes Plugin) fest:
> 
>     * Festlegung der gewünschten Einstellungen
> 
>     * dann auf das Zahnradsymbol oben links im Kartenfenster klicken.
> 
>     * Drücken Sie die Sterntaste unten links im Dialogfeld Basic Feature
> 
>     * Drücken Sie die Plustaste
> 
>     * Gruppe auf Standard setzen
> 
>     * Set Beschreibung zum Standard
> 
>     * Presse OK

> What do you mean by 'restarting'? By exiting SDRangel and starting it again? Or closing the map and opening a new map?
>
> In the former case, the settings should be saved. In the latter case they wont [...]

select Sampling Device RTL Stick ADD Available features (MAP) MAP Show Settings Dialog MAP Items first column (enab e) "ACARSDemod down to Wapoints" The enable disable function does not save for the next start. The MAP Items must always be set to me again and again enable disabe I tested it with Linux MINT Windows 10 11, no saving possible.

KanotixPinguin, 2025-03-01:
> What is your client program? What messages do you see in the SDRangel debug log file? (I.e. might be easier to run the server via the GUI to debug what is going on).

my client program is SDRangel the latest version tested under Linux Mint and Windows 10 11x64 I'll describe it again My Server Remote TCP sink from ADD feature Remote TCP sink. Address 0.0.0.0 Port 1234. the server IP address 192.168.178.88 SR 02.000.000 IQ-8bits Protocol SDRangel. Puplic Directory LIST on Server everything filled in. my client SDRangel latest version tested with Linux Mint Windoiws 10 and 11 The connection remote TCPInput from sampling device to my other PC in the network and also from the outside the IP of the provider and my dynamic IP address (works, there are no problems). Only after exiting the client on PC again tested under Linux Mint Windows 10 and 11 (no longer working) sometimes the trick from time to time when remote TCPInput play and then IP 192.168.178.55 or dynamic address port 1234 click the IP button down, make 127.0.0.1 to 192.168.178.55 connect ok. But not always. I then have to stop the server on 192.168.178.88. Then I start the server Remote TCP sink as a feature, everything new with RTL stick and all futures. Then I can connect on 192.168.178.55 or network computer or even outside. But even there I have to enter 127.0.0.1 switch to 192.168.178.88 under the IP button. It seems to me that the rights to save are not given. I tested everything on Linux and Windows 10 and 11

[image]

[image]

[image]

KanotixPinguin, 2025-03-01:
> > If I may point out, the scanner function is very slow
> 
> Please give a more detailed description of what settings you are using and how fast you think it should be.

Device RTL-SDR Channels Scanner. The scanning of the frequencies of the stored frequencies is a bit slow, the milliseconds you could be between 60 and 20 milliseconds, the pause 8.0 until it continues scanning.

closed 2025-04-17 by github-actions[bot]