f4exb/sdrangel #2411: Can't Download Nav Files for VOR Localizer - .xml's no longer exist (only .mbtiles files)

MeadeRobert, 2025-03-02:
2025-03-01 21:35:30.312 (I) SDRangel 7.22.6 Qt 6.8.2 64b x86_64 KDE Flatpak runtime DSP Rx:24b Tx:16b PID 2
2025-03-01 21:35:32.537 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ad_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ad_nav.xml - server replied: "
2025-03-01 21:35:32.898 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ae_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ae_nav.xml - server replied: "
2025-03-01 21:35:33.607 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ai_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ai_nav.xml - server replied: "
2025-03-01 21:35:34.179 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/am_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/am_nav.xml - server replied: "
2025-03-01 21:35:34.539 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/an_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/an_nav.xml - server replied: "
2025-03-01 21:35:34.673 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ao_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ao_nav.xml - server replied: "
2025-03-01 21:35:36.862 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ax_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ax_nav.xml - server replied: "
2025-03-01 21:35:37.032 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/az_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/az_nav.xml - server replied: "
2025-03-01 21:35:37.917 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bd_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bd_nav.xml - server replied: "
2025-03-01 21:35:38.420 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bf_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bf_nav.xml - server replied: "
2025-03-01 21:35:39.261 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bi_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bi_nav.xml - server replied: "
2025-03-01 21:35:39.448 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bj_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bj_nav.xml - server replied: "
2025-03-01 21:35:39.754 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bl_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bl_nav.xml - server replied: "
2025-03-01 21:35:40.514 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bq_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bq_nav.xml - server replied: "
2025-03-01 21:35:41.142 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bt_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bt_nav.xml - server replied: "
2025-03-01 21:35:41.266 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bv_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/bv_nav.xml - server replied: "
2025-03-01 21:35:41.582 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/by_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/by_nav.xml - server replied: "
2025-03-01 21:35:42.301 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cd_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cd_nav.xml - server replied: "
2025-03-01 21:35:42.434 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cf_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cf_nav.xml - server replied: "
2025-03-01 21:35:42.568 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cg_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cg_nav.xml - server replied: "
2025-03-01 21:35:42.887 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ci_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ci_nav.xml - server replied: "
2025-03-01 21:35:43.515 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cm_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cm_nav.xml - server replied: "
2025-03-01 21:35:44.864 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cv_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/cv_nav.xml - server replied: "
2025-03-01 21:35:46.211 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/dj_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/dj_nav.xml - server replied: "
2025-03-01 21:35:48.360 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/er_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/er_nav.xml - server replied: "
2025-03-01 21:35:48.853 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/et_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/et_nav.xml - server replied: "
2025-03-01 21:35:50.530 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ga_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/ga_nav.xml - server replied: "
2025-03-01 21:35:51.678 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/gh_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/gh_nav.xml - server replied: "
2025-03-01 21:35:52.199 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/gm_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/gm_nav.xml - server replied: "
2025-03-01 21:35:52.331 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/gn_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/gn_nav.xml - server replied: "
2025-03-01 21:35:52.638 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/gq_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489d [... cut ...]
2025-03-01 21:36:20.531 (C) HttpDownloadManager: Download of  "https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/zw_nav.xml"  failed:  "Error transferring https://storage.googleapis.com/29f98e10-a489-4c82-ae5e-489dbcd4912f/zw_nav.xml - server replied: "
2025-03-01 21:36:25.026 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_currentItemChanged(QTreeWidgetItem*,QTreeWidgetItem*)
2025-03-01 21:36:25.026 (W) QMetaObject::connectSlotsByName: No matching signal for on_presetTree_itemActivated(QTreeWidgetItem*,int)
2025-03-01 21:36:44.945 (I) libpng warning: iCCP: known incorrect sRGB profile

srcejon, 2025-03-02:
What country are you in?

I'd expect some downloads to fail, as there isn't an xml for all countries. I don't see errors for the larger countries, so download is probably ok.

MeadeRobert, 2025-03-02:
USA. The bigger problem I'm having is that the VOR localizer won't display any map at all regardless of what position I set in the preferences. Also, it's unclear form the docs whether the morse ID needs to be functioning correctly to allow the VOR bearing to propagate over to the VOR localizer.

srcejon, 2025-03-02:
Can you add a full screenshot please?

MeadeRobert, 2025-03-02:
[image]

MeadeRobert, 2025-03-02:
Strictly speaking my VOR isn't in the database, but is there a way to manually add it so it will display in the localizer?

srcejon, 2025-03-02:
The map not working is probably because you are using a flatpak / qt 6 build. Should work on qt 5.

You can add the VOR manually to the download files. Look in us_nav.xml

closed 2025-04-17 by github-actions[bot]