f4exb/sdrangel #2415: Windows build failing

ikjordan, 2025-03-07:
The Windows build on master is currently failing. Looking at the pull request logs it apopears that the Windows build has been failing since  #2410. 

I was attempting to test the fix applied in #2414, but also could not build locally. I was previously building for Windows successfully as I was helping with the investigations for #2414

srcejon, 2025-03-07:
Looks like LimeSuite is failing to build with the most recent version of Visual Studio.

srcejon, 2025-03-07:
Looks like #include <chrono> needs to be added to a few of the LimeSuite files.

srcejon, 2025-03-07:
Have added: https://github.com/myriadrf/LimeSuite/issues/414

ikjordan, 2025-03-07:
I applied the changes for LimeSuite. 
The build then failed later, but completed successfully when I also added `#include <chrono>` to `build\external\dab\src\dab\library\includes\support\dab-semaphore.h`
The good news is that the build then ran successfully, and I could use my hackRF to verify that https://github.com/f4exb/sdrangel/pull/2414 is fixed

commit 825c1d07 (2025-03-07): Update LimeSuite to latest version, to fix #2415
closed 2025-03-07 by f4exb