f4exb/sdrangel #2424: LimeSDR RX and TX frequencies locked despite using separate VCOs

lukolszewski, 2025-03-25:
Hi,

Another similar issue created for HackRF had a comment this behaviour is caused by the fact HackRF has a single VCO. This is not the case with LimeSDR. However it is not possible to set as different frequency for RX and TX and have these remain such (even if one adds RX from channel 0 and TX from channel 1 - LimeUSB has 2x2 MIMO so it is capable of RX/TX on 4 separate frequencies).

First when using RX from Lime ID 0 and TX from Lime ID 1:
When one starts the TX sink (while the RX source is running on a different frequency) it seems to run fine, but then when the TX sink is stopped the RX source has its frequency updated to the TX frequency without the GUI reflecting it. In the GUI it is as if nothing changed, but it has. The spectrum is shifted.

Then if using the RX and TX from the same Lime ID the frequencies get synced when the second sink or source is started.

To be able to use MIMO the frequencies have to remain on what they were set. Please advise which part of the code is doing it.

srcejon, 2025-03-26:
> LimeUSB has 2x2 MIMO so it is capable of RX/TX on 4 separate frequencies).

The LMS7002M datasheet says: "Both transmitters share one PLL and both receivers share another"

lukolszewski, 2025-03-26:
> > LimeUSB has 2x2 MIMO so it is capable of RX/TX on 4 separate frequencies).
> 
> The LMS7002M datasheet says: "Both transmitters share one PLL and both receivers share another"

Yes, I was mistaken, there are 2 VCOs and one clock for all of them. This allows use on two separate frequencies as long as they can share the same clock. One for both TX and one for both RX.

As far as I can tell there should be no reason to lock TX and RX to the same frequency.

Regarding the previously mentioned buddies concept. It appears to simply be, if there is more than one sink/sourcve on the same interface they are set up as "buddies". Like on the same USB port for example. I tried removing the code that adds to the buddy list and sdrangel couldn't talk to more than one device via one USB port. So I reverted the change.

In the meantime I saw in applySettings function in limesdrinput.cpp line 802 there is some logic that based on what is being updated sends a message to "buddies" either TX buddies or RX buddies to update themselves.  This is a good idea for many things like sample rate and so on. In fact if one updates the center frequency this is not enough to trigger all buddy update (only RX buddies if this is RX or only TX buddies if this is TX - this makes sense because one VCO is shared beyweenm RX and another VCO is shared between TX).

However, there is also a "force parameter" that appears to force an update to all buddies. And when the start button is pressed in limesdrinput.cpp line 438 it triggers an all settings apply that includes this force parameter. No idea why.

I commented out that line and sdrangel appears to be running fine and the frequencies between RX and TX are no longer locked. I did experience one crash, but I doubt it is related. If anyone cazn shed some light on this it would be great.

lukolszewski, 2025-03-26:
This diff resolves the problem for me (it actually reverts to an older code). Now the TX and RX clocks are independent. 

```
diff --git a/plugins/samplesink/limesdroutput/limesdroutput.cpp b/plugins/samplesink/limesdroutput/limesdroutput.cpp
index 9551a18df..b8604d8be 100644
--- a/plugins/samplesink/limesdroutput/limesdroutput.cpp
+++ b/plugins/samplesink/limesdroutput/limesdroutput.cpp
@@ -807,7 +807,9 @@ bool LimeSDROutput::applySettings(const LimeSDROutputSettings& settings, const Q
     if (settingsKeys.contains("devSampleRate")
        || settingsKeys.contains("log2HardInterp") || force)
     {
-        forwardChangeAllDSP = true; //m_settings.m_devSampleRate != settings.m_devSampleRate;
+        //forwardChangeAllDSP = true; //m_settings.m_devSampleRate != settings.m_devSampleRate;
+        forwardChangeTxDSP  = m_settings.m_log2HardInterp != settings.m_log2HardInterp;
+        forwardChangeAllDSP = m_settings.m_devSampleRate != settings.m_devSampleRate;

         if (m_deviceShared.m_deviceParams->getDevice())
         {
diff --git a/plugins/samplesource/limesdrinput/limesdrinput.cpp b/plugins/samplesource/limesdrinput/limesdrinput.cpp
index 62d1a6efc..63b7f3dcc 100644
--- a/plugins/samplesource/limesdrinput/limesdrinput.cpp
+++ b/plugins/samplesource/limesdrinput/limesdrinput.cpp
@@ -969,7 +969,9 @@ bool LimeSDRInput::applySettings(const LimeSDRInputSettings& settings, const QLi
     if (settingsKeys.contains("devSampleRate")
        || settingsKeys.contains("log2HardDecim") || force)
     {
-        forwardChangeAllDSP = true; //m_settings.m_devSampleRate != settings.m_devSampleRate;
+        //forwardChangeAllDSP = true; //m_settings.m_devSampleRate != settings.m_devSampleRate;
+        forwardChangeRxDSP  = m_settings.m_log2HardDecim != settings.m_log2HardDecim;
+        forwardChangeAllDSP = m_settings.m_devSampleRate != settings.m_devSampleRate;

         if (m_deviceShared.m_deviceParams->getDevice() && m_channelAcquired)
         {
```

But sometimes one may want them locked.... I think it would be really nice to have a tick box in the gui one could tick to lock it together or split. 

It is up to the author to decice which way to go with this. I'd love to have an option.

lukolszewski, 2025-03-27:
PR created, https://github.com/f4exb/sdrangel/pull/2425

closed 2025-05-11 by github-actions[bot]