f4exb/sdrangel #2427: PlutoSDR frequency expansion mod bandwidth

madmike-514, 2025-04-10:
Hello, sorry if this is not the best place for this but I am having trouble using the expanded bandwidth of 56MHz with my PlutoSDR.
The PlutoSDR has a default bandwidth of 20 MHz and a frequency range of 325 MHz to 3.8 GHz.
With the mod it has a frequency range of 70 MHz to 6 GHz with a maximum bandwidth of 56 MHz.
Could I just modify these lines?
const uint32_t DevicePlutoSDR::srLowLimitFreq  = (25000000U/12U)+3U; // 25/12 MS/s without FIR interpolation/decimation (+3 so it is the next multiple of 4)
const uint32_t DevicePlutoSDR::srHighLimitFreq = 20000000U;     // 20 MS/s: take AD9363 speces
Is there a potential for damage to the device doing this?
Good luck with all your projects! Thanks for any help!

f4exb, 2025-04-14:
It would be better to use the libiio interface provided there is a function returning the actual maximum sample rate and is reliable (I did not check).

madmike-514, 2025-04-15:
Thanks for the response sir. I think the USB 2.0 will limit getting data off the device anyway, I can see about 5 or 6 MHz max even after compiling on a much faster machine.
sdrangel has been the most fun I have had with a piece of software in years. Thanks.

closed 2025-06-01 by github-actions[bot]