f4exb/sdrangel #2430: Trying to build on m2 mac fails

mattsoftware, 2025-04-13:
I believe the homebrew installed boost libraries are too new for the build. I'm not sure how to get past this.

```
[  0%] Building CXX object lib/deps/rpclib/CMakeFiles/uhd_rpclib.dir/lib/rpc/server.cc.o
cd /xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd-build/lib/deps/rpclib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_STRING_VIEW -DBOOST_ERROR_CODE_HEADER_ONLY -DHAVE_CONFIG_H -DUHD_LOG_CONSOLE_COLOR -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_MIN_LEVEL=1 -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd-build/include -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/include -I/opt/homebrew/include -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/include -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd-build/lib/include -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/flatbuffers/include -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd-build/lib/ic_reg_maps -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/convert -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd-build/lib/convert -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/usrp -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/usrp/common/ad9361_driver -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/usrp/common -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd-build/lib/transport/nirio/lvbitx -I/opt/homebrew/Cellar/libusb/1.0.27/include/libusb-1.0 -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include -I/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include/rpc -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG -std=c++14 -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk -fPIC   -Wall -Wextra -Wsign-compare -MD -MT lib/deps/rpclib/CMakeFiles/uhd_rpclib.dir/lib/rpc/server.cc.o -MF CMakeFiles/uhd_rpclib.dir/lib/rpc/server.cc.o.d -o CMakeFiles/uhd_rpclib.dir/lib/rpc/server.cc.o -c /xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/lib/rpc/server.cc
In file included from /xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/lib/rpc/server.cc:14:
In file included from /xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include/rpc/detail/server_session.h:14:
/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include/rpc/detail/async_writer.h:22:31: error: no type named 'io_service' in namespace 'boost::asio'
   22 |     async_writer(boost::asio::io_service *io,
      |                  ~~~~~~~~~~~~~^
/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include/rpc/detail/async_writer.h:72:18: error: no member named 'io_service' in namespace 'boost::asio'
   72 |     boost::asio::io_service::strand write_strand_;
      |     ~~~~~~~~~~~~~^
/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include/rpc/detail/async_writer.h:30:19: error: no matching member function for call to 'shared_from_this'
   30 |         auto self(shared_from_this());
      |                   ^~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk/usr/include/c++/v1/__memory/shared_ptr.h:1530:41: note: candidate function not viable: no known conversion from 'rpc::detail::async_writer' to 'std::enable_shared_from_this<rpc::detail::async_writer>' for object argument
 1530 |   _LIBCPP_HIDE_FROM_ABI shared_ptr<_Tp> shared_from_this() { return shared_ptr<_Tp>(__weak_this_); }
      |                                         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk/usr/include/c++/v1/__memory/shared_ptr.h:1531:47: note: candidate function not viable: no known conversion from 'rpc::detail::async_writer' to 'const std::enable_shared_from_this<rpc::detail::async_writer>' for object argument
 1531 |   _LIBCPP_HIDE_FROM_ABI shared_ptr<_Tp const> shared_from_this() const { return shared_ptr<const _Tp>(__weak_this_); }
      |                                               ^
In file included from /xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/lib/rpc/server.cc:14:
/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include/rpc/detail/server_session.h:25:46: error: no type named 'io_service' in namespace 'boost::asio'
   25 |     server_session(server *srv, boost::asio::io_service *io,
      |                                 ~~~~~~~~~~~~~^
/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include/rpc/detail/server_session.h:37:18: error: no type named 'io_service' in namespace 'boost::asio'
   37 |     boost::asio::io_service *io_;
      |     ~~~~~~~~~~~~~^
/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/include/rpc/detail/server_session.h:38:18: error: no member named 'io_service' in namespace 'boost::asio'
   38 |     boost::asio::io_service::strand read_strand_;
      |     ~~~~~~~~~~~~~^
/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/lib/rpc/server.cc:69:5: error: unknown type name 'io_service'
   69 |     io_service io_;
      |     ^
/xxxxxxxxxxxxxxxxxxxxx/sdrangel/build/external/uhd/src/uhd/host/lib/deps/rpclib/lib/rpc/server.cc:28:48: error: no member named 'from_string' in 'boost::asio::ip::address'
   28 |                     tcp::endpoint(ip::address::from_string(address), port)),
      |                                   ~~~~~~~~~~~~~^
8 errors generated.
make[5]: *** [lib/deps/rpclib/CMakeFiles/uhd_rpclib.dir/lib/rpc/server.cc.o] Error 1
make[4]: *** [lib/deps/rpclib/CMakeFiles/uhd_rpclib.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [external/uhd/src/uhd-stamp/uhd-build] Error 2
make[1]: *** [external/CMakeFiles/uhd.dir/all] Error 2
make: *** [all] Error 2
```

I tried to get as much information for building on mac as I could from the github workflow file which got me this far. Chat GPT says I need to downgrade the boost library but it seems there is no compatible version I can install (using brew at least).

Thanks for any help on this.

srcejon, 2025-04-13:
Your problem is in UHD code rather than SDRangel itself. Do you need to use the UHD drivers (for USRP SDRs)? 

If not, run cmake with -DENABLE_USRP=OFF

mattsoftware, 2025-04-14:
Thank you @srcejon . That got me further. My final command was ... ```cmake .. -DCMAKE_VERBOSE_MAKEFILE=ON -DENABLE_QT6=ON -DCMAKE_BUILD_TYPE=Release -DDEBUG_OUTPUT=ON -DENABLE_USRP=OFF -DENABLE_MIRISDR=OFF -DBUILD_SERVER=OFF -DENABLE_EXTERNAL_LIBRARIES=ON -DBUNDLE=ON -DPKG_CONFIG_USE_CMAKE_PREFIX_PATH=TRUE -DPYTHON_EXECUTABLE=/xxxxxxxxxx/.pyenv/shims/python -DENABLE_CHANNELRX_REMOTESINK=OFF -DENABLE_CHANNELRX_DEMODNAVTEX=OFF -DENABLE_CHANNELTX_REMOTESOURCE=OFF -DENABLE_CHANNELMIMO=OFF```

which got me succesfully compiling it at least. When I run the ./sdrangel I get another error ```dyld[80052]: Library not loaded: @rpath/libserialdv.dylib
  Referenced from: <B71A2C67-3C9F-341E-BA1D-75365C99C3F0> /Users/matt/Github/MSW/sdrangel/build/sdrangel
  Reason: tried: '/opt/homebrew/lib/libserialdv.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/lib/libserialdv.dylib' (no such file), '/Users/matt/Github/MSW/sdrangel/build/lib/libserialdv.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/matt/Github/MSW/sdrangel/build/lib/libserialdv.dylib' (no such file), '/opt/homebrew/lib/libserialdv.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/lib/libserialdv.dylib' (no such file), '/Users/matt/Github/MSW/sdrangel/build/lib/libserialdv.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/matt/Github/MSW/sdrangel/build/lib/libserialdv.dylib' (no such file)
zsh: abort      ./sdrangel```

So I ran the `make package` command which built a dmg file which of course is not signed correctly. I dont want to sound like an idiot here, but is there an easy way to just do a build and run it locally so I can start hacking on some code? Any hints on doing this would be fantastic. Even if I can remove the certificate when building the application would be a help.

Thanks again.

srcejon, 2025-04-14:
We don't sign the package - but you can't still install and run it anyway.

mattsoftware, 2025-04-14:
Thats interesting. I'm getting a code signature invalid when I try and run the .app from the .dmg ... 

```-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SDRangel [87196]
Path:                  /Volumes/*/SDRangel.app/Contents/MacOS/sdrangel
Identifier:            SDRangel
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-04-15 08:47:38.6585 +1000
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        7E597C22-4BFC-D94B-C960-B0E1EF6BF560

Sleep/Wake UUID:       C1549F7B-BB6F-4DEB-AD2F-1BE36C15D67C

Time Awake Since Boot: 43000 seconds
Time Since Wake:       1608 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x000000010529c000
Exception Codes:       0x0000000000000032, 0x000000010529c000

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page

VM Region Info: 0x10529c000 is in 0x10529c000-0x105640000;  bytes after start: 0  bytes before end: 3817471
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      mapped file                 104d74000-10529c000    [ 5280K] r--/rwx SM=COW  Object_id=3ebd427d
--->  mapped file                 10529c000-105640000    [ 3728K] r-x/rwx SM=COW  Object_id=3ebd427d
      VM_ALLOCATE (reserved)      105640000-1057c4000    [ 1552K] rw-/rwx SM=NUL  reserved VM address space (unallocated)

Thread 0 Crashed:
0   dyld                          	       0x185b6eef0 dyld3::MachOFile::isMachO(Diagnostics&, unsigned long long) const + 40```

mattsoftware, 2025-04-15:
Okay, next experiment was to basically turn off all the feature flags and only include the one that I want to debug. Still getting an error, this time with flac. I must say I am not enjoying trying to work with this project at the code level....

'''[ 95%] Linking CXX shared module ../../../lib/plugins/libinputtcpremote.dylib
cd /Users/matt/Github/MSW/sdrangel/build/plugins/samplesource/remotetcpinput && /opt/homebrew/bin/cmake -E cmake_link_script CMakeFiles/inputtcpremote.dir/link.txt --verbose=1
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -O3 -DNDEBUG -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk -bundle -Wl,-headerpad_max_install_names -o ../../../lib/plugins/libinputtcpremote.dylib CMakeFiles/inputtcpremote.dir/inputtcpremote_autogen/mocs_compilation.cpp.o CMakeFiles/inputtcpremote.dir/remotetcpinputtcphandler.cpp.o CMakeFiles/inputtcpremote.dir/remotetcpinput.cpp.o CMakeFiles/inputtcpremote.dir/remotetcpinputsettings.cpp.o CMakeFiles/inputtcpremote.dir/remotetcpinputwebapiadapter.cpp.o CMakeFiles/inputtcpremote.dir/remotetcpinputplugin.cpp.o CMakeFiles/inputtcpremote.dir/remotetcpinputgui.cpp.o -F/opt/homebrew/lib  -Wl,-rpath,/opt/homebrew/lib ../../../lib/libsdrbase.a ../../../lib/libsdrgui.a ../../../lib/libswagger.a ../../../external/zlib/lib/libz.dylib ../../../lib/libsdrbase.a /opt/homebrew/lib/QtWebSockets.framework/Versions/A/QtWebSockets ../../../lib/libswagger.a ../../../external/opus/lib/libopus.dylib /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk/usr/lib/libm.tbd ../../../external/fftw3f/src/fftw3f-build/.libs/libfftw3f.a ../../../external/serialdv/src/serialdv-build/libserialdv.dylib ../../../lib/libhttpserver.a ../../../lib/libqrtplib.a /opt/homebrew/lib/QtOpenGLWidgets.framework/Versions/A/QtOpenGLWidgets /opt/homebrew/lib/QtWidgets.framework/Versions/A/QtWidgets /opt/homebrew/lib/QtMultimedia.framework/Versions/A/QtMultimedia /opt/homebrew/lib/QtPositioning.framework/Versions/A/QtPositioning ../../../lib/liblogging.a /opt/homebrew/lib/QtNetwork.framework/Versions/A/QtNetwork /opt/homebrew/lib/QtOpenGL.framework/Versions/A/QtOpenGL -framework OpenGL /opt/homebrew/lib/QtStateMachine.framework/Versions/A/QtStateMachine /opt/homebrew/lib/QtGui.framework/Versions/A/QtGui -framework AGL -framework AppKit -framework OpenGL -framework ImageIO -framework Metal /opt/homebrew/lib/QtCore.framework/Versions/A/QtCore -framework IOKit -framework DiskArbitration -framework UniformTypeIdentifiers
ld: warning: ignoring duplicate libraries: '../../../lib/libsdrbase.a', '../../../lib/libswagger.a'
Undefined symbols for architecture arm64:
  "_FLAC__stream_decoder_delete", referenced from:
      RemoteTCPInputTCPHandler::cleanup() in remotetcpinputtcphandler.cpp.o
  "_FLAC__stream_decoder_init_stream", referenced from:
      RemoteTCPInputTCPHandler::processMetaData() in remotetcpinputtcphandler.cpp.o
  "_FLAC__stream_decoder_new", referenced from:
      RemoteTCPInputTCPHandler::processMetaData() in remotetcpinputtcphandler.cpp.o
  "_FLAC__stream_decoder_process_single", referenced from:
      RemoteTCPInputTCPHandler::processCommands() in remotetcpinputtcphandler.cpp.o
ld: symbol(s) not found for architecture arm64
c++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/plugins/libinputtcpremote.dylib] Error 1
make[1]: *** [plugins/samplesource/remotetcpinput/CMakeFiles/inputtcpremote.dir/all] Error 2
make: *** [all] Error 2'''

I have added this to my build command, but it didn't make a difference....
    -DFLAC_INCLUDE_DIR=/opt/homebrew/Cellar/flac/1.5.0/include \
    -DFLAC_LIBRARY=/opt/homebrew/Cellar/flac/1.5.0/lib/libFLAC.dylib \

Hoping to get any suggestions on how to build this project so I can do some real coding. Seem to be wasting a bunch of time simply trying to get it to the point where I can actually look at the code 👎

mattsoftware, 2025-04-15:
Another small update. I had to set  `-DENABLE_CHANNELRX_REMOTETCPSINK=ON` to get rid of the above error. Still results in a binary that does'nt run on my machine.

srcejon, 2025-04-15:
> I'm getting a code signature invalid when I try and run the .app from the .dmg .

Did you try:

"The first time you run SDRangel, you must hold the Control key and click the SDRangel icon and select Open from the menu, then press the Open button in the dialog that asks if you are sure you want to open it."

mattsoftware, 2025-04-18:
Yea, i understand all the security requirements around mac binaries. That is no the issue. After further digging at lot of the errors seem to stem from my locally installed version of boost. No matter what I try and do some of the dependencies will always choose it over trying to build it. I had to remove the library from homebrew installation and then cmake seems to build it as a dependency which is good. However then i get a flac issue, so I uninstall that, and now I have more errors. I've tried about 100 different things and I'm getting to the end of my tether trying to build the library, at least on mac. I will have to attempt on linux and hope that goes well as I need a way to build and try to debug other issues I am having.

If there is a bullet proof command I can run to build on mac that would be awesome. I am having no luck trying to get the build to compile, and i've removed as must as I can from the build to a) make it faster, but also b) try and compile less so I get less issues.

Either way I will keep trying, but i'm not sure how much more I can fail before giving up unfortunately :)

mattsoftware, 2025-04-21:
I cannot get it working on mac. At the moment I am relying on the 2.5 hour github actions to compile. Closing for now and i'll attempt to build again in the future once I understand more about the build process and how to fix the issues I am up against.

closed 2025-04-21 by mattsoftware