f4exb/sdrangel #2434: System load issue when logging Demod plugin data via REST API on Raspberry Pi

msaideroglu, 2025-04-18:
Hi,

I'm encountering system stability issues when using SDRangel headless (`sdrangelsrv`) on a Raspberry Pi (Raspberry Pi 4, 8GB RAM). My setup includes:

- RTL-SDR source with 2.4 MSps
- FileSink plugin for IQ recording
- Three `Demod` plugins configured for simultaneous demodulation
- A Python script polling REST API `/report` endpoints of the Demod channels at 1 Hz
- Data is logged locally on the Pi while SDRangel is recording

After a few minutes, the Raspberry Pi becomes unresponsive:
- CPU load spikes significantly
- The system overheats
- Eventually, I lose the SSH connection due to CPU overload or thermal shutdown

This seems to be caused by the combination of high polling rate and JSON serialization on top of simultaneous real-time processing and file I/O.

1. Is there a more CPU-efficient way to log Demod plugin outputs?  
   - For example: having demod plugins write data directly to disk or Reverse API?

2. Would it be possible for the plugin to expose data via UDP or shared memory to avoid JSON overhead?

3. Would it help if API polling was offloaded to a separate device over LAN, so Raspberry only handles SDRangel runtime?

Any suggestions or best practices would be appreciated. Thanks!

srcejon, 2025-04-18:
Which demod?

msaideroglu, 2025-04-20:
There are 3 VOR Demodulators which Rest API is logging VOR radials and signal Power(dB).

msaideroglu, 2025-04-21:
Here is my test scenario: I start the RTL-SDR source and begin recording using the File Sink. In this setup, there are no issues with CPU load. However, when I run a Python script that sends API requests to log the VOR radials from 3 different VOR demodulator channels, the system starts to exhibit high CPU usage or thermal issues. After 10–15 minutes, this leads to system instability or a complete freeze on the Raspberry Pi. Maybe 3 VOR Demodulators' themself consume whole CPU load. 

How I can understand CPU load consumed is that I'm getting outputs in sdrangelsrv's terminal like this:
2024-01-28 08:59:59.164 (C) SampleSinkFifo::write: (SDRPlayV3) 2306 messages dropped
2024-01-28 08:59:59.165 (C) SampleSinkFifo::write: (SDRPlayV3) overflow - dropping 1024 samples
2024-01-28 08:59:59.165 (C) SampleSinkFifo::write: (SDRPlayV3) overflow - dropping 1024 samples

srcejon, 2025-04-21:
> Maybe 3 VOR Demodulators' themself consume whole CPU load.

Well, you can test that by running the demods without your python script.

srcejon, 2025-04-21:
> * RTL-SDR source with 2.4 MSps

Do you need to run at this sample rate? What frequencies are the VORs at?

msaideroglu, 2025-04-21:
Without Python script, it runs with no problem. When Python starts to run, the problem occurs in 10 minutes. It makes me think that Cpu load gets full and freezes.

Yes I nee 2.4mhz. Vor are in 112.1, 113.1 and 114.3. So A 2.4mhz bw can include all of them.

srcejon, 2025-04-21:
Do you have a complete Python script I can try?

closed 2025-04-24 by msaideroglu