f4exb/sdrangel #2436: When building getitng compatiblity with CMake < 3.5 errors

mattsoftware, 2025-04-19:
Trying to build using github ci as well as local mac, I'm starting to see these errors...

```
-- Already at requested tag: v1.1.4
[14/2600] Performing update step for 'mbelib'
-- Already at requested ref: fe83b32c6a60cdd7bce8cecf3c7a0b9ec87a7667
[15/2600] No patch step for 'serialdv'
[16/2600] Performing configure step for 'serialdv'
FAILED: external/serialdv/src/serialdv-stamp/serialdv-configure D:/a/sdrangel/sdrangel/build/external/serialdv/src/serialdv-stamp/serialdv-configure 
C:\Windows\system32\cmd.exe /C "cd /D D:\a\sdrangel\sdrangel\build\external\serialdv\src\serialdv-build && "C:\Program Files\CMake\bin\cmake.exe" -DBUILD_TOOL=OFF -DCMAKE_RUNTIME_OUTPUT_DIRECTORY=D:/a/sdrangel/sdrangel/build/bin -DCMAKE_RUNTIME_OUTPUT_DIRECTORY_DEBUG=D:/a/sdrangel/sdrangel/build/bin -DCMAKE_RUNTIME_OUTPUT_DIRECTORY_RELEASE=D:/a/sdrangel/sdrangel/build/bin -DCMAKE_RUNTIME_OUTPUT_DIRECTORY_RELWITHDEBINFO=D:/a/sdrangel/sdrangel/build/bin -DCMAKE_ARCHIVE_OUTPUT_DIRECTORY=D:/a/sdrangel/sdrangel/build/lib -DCMAKE_ARCHIVE_OUTPUT_DIRECTORY_DEBUG=D:/a/sdrangel/sdrangel/build/lib -DCMAKE_ARCHIVE_OUTPUT_DIRECTORY_RELEASE=D:/a/sdrangel/sdrangel/build/lib -DCMAKE_ARCHIVE_OUTPUT_DIRECTORY_RELWITHDEBINFO=D:/a/sdrangel/sdrangel/build/lib -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=D:/a/sdrangel/sdrangel/build/lib -DCMAKE_LIBRARY_OUTPUT_DIRECTORY_DEBUG=D:/a/sdrangel/sdrangel/build/lib -DCMAKE_LIBRARY_OUTPUT_DIRECTORY_RELEASE=D:/a/sdrangel/sdrangel/build/lib -DCMAKE_LIBRARY_OUTPUT_DIRECTORY_RELWITHDEBINFO=D:/a/sdrangel/sdrangel/build/lib -DCMAKE_POLICY_DEFAULT_CMP0091:STRING=NEW -DCMAKE_MSVC_RUNTIME_LIBRARY=MultiThreadedDLL -DCMAKE_BUILD_TYPE=Release -DCMAKE_POLICY_DEFAULT_CMP0069:STRING=NEW -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=ON "-DCMAKE_C_COMPILER=C:/Program Files/Microsoft Visual Studio/2022/Enterprise/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe" "-DCMAKE_CXX_COMPILER=C:/Program Files/Microsoft Visual Studio/2022/Enterprise/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe" "-DCMAKE_LINKER=C:/Program Files/Microsoft Visual Studio/2022/Enterprise/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/link.exe" "-DCMAKE_AR=C:/Program Files/Microsoft Visual Studio/2022/Enterprise/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/lib.exe" -DCMAKE_CXX_STANDARD=17 -DCMAKE_CXX_STANDARD_REQUIRED=ON -DCMAKE_CXX_EXTENSIONS=OFF -GNinja -S D:/a/sdrangel/sdrangel/build/external/serialdv/src/serialdv -B D:/a/sdrangel/sdrangel/build/external/serialdv/src/serialdv-build && "C:\Program Files\CMake\bin\cmake.exe" -E touch D:/a/sdrangel/sdrangel/build/external/serialdv/src/serialdv-stamp/serialdv-configure"
  CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
[17/2600] No patch step for 'mbelib'
[18/2600] Building CXX object httpserver\CMakeFiles\httpserver.dir\httpserver_autogen\mocs_compilation.cpp.obj
[19/2600] Performing download step (git clone) for 'pthreads4w'
Cloning into 'pthreads4w'...
HEAD is now at 9808f0b Update for release
[20/2600] Performing download step (git clone) for 'libxml2'
Cloning into 'libxml2'...
HEAD is now at 223cb03a Release v2.10.4
[21/2600] Performing download step (git clone) for 'limesuite'
Cloning into 'limesuite'...
HEAD is now at b39cb61e Add more missing MSVC chrono include
ninja: build stopped: subcommand failed.```

I tried to push the last successful ci build from master which was 81bae39. I pushed to the mattsoftware/sdrangel to master and getting these errors. I am finding this project increasingly impossible to build and falling back to the github CI pipeline doens't seem to be working either.

srcejon, 2025-04-21:
This should have been fixed a couple of weeks ago: 

https://github.com/f4exb/sdrangel/commit/c7e2a7890329100533dee9e514dac28c176700cb
https://github.com/f4exb/serialDV/commit/217f322b8c51a0182a2bd83d27f4b1c913cfff26

You need to check why you are seeing:

Already at requested tag: v1.1.4

Rather than v1.1.5.

mattsoftware, 2025-04-22:
Sorry, I think I was trying to build on an older branch.

closed 2025-04-22 by mattsoftware