f4exb/sdrangel #2441: Windows CI-Pipeline failing with python error.

mattsoftware, 2025-04-22:
Current master is failing the windows pipeline with this error...

[Windows-x64](https://github.com/mattsoftware/sdrangel/actions/runs/14604910965/job/40971669840#step:7:265)
Error: The process 'C:\hostedtoolcache\windows\Python\3.9.13\x64\python.exe' failed with exit code 1

mattsoftware, 2025-04-22:
```Successfully installed aqtinstall-3.2.1 beautifulsoup4-4.13.4 bs4-0.0.2 certifi-2025.1.31 charset-normalizer-3.4.1 defusedxml-0.7.1 humanize-4.12.2 idna-3.10 patch-ng-1.18.1 requests-2.32.3 semantic-version-2.10.0 soupsieve-2.7 typing-extensions-4.13.2 urllib3-2.4.0
C:\hostedtoolcache\windows\Python\3.9.13\x64\python.exe -m aqt version
Traceback (most recent call last):
  File "C:\hostedtoolcache\windows\Python\3.9.13\x64\lib\runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "C:\hostedtoolcache\windows\Python\3.9.13\x64\lib\runpy.py", line 147, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "C:\hostedtoolcache\windows\Python\3.9.13\x64\lib\runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "C:\hostedtoolcache\windows\Python\3.9.13\x64\lib\site-packages\aqt\__init__.py", line 30, in <module>
    from aqt.installer import Cli
  File "C:\hostedtoolcache\windows\Python\3.9.13\x64\lib\site-packages\aqt\installer.py", line 45, in <module>
    from aqt.archives import QtArchives, QtPackage, SrcDocExamplesArchives, ToolArchives
  File "C:\hostedtoolcache\windows\Python\3.9.13\x64\lib\site-packages\aqt\archives.py", line 32, in <module>
    from aqt.helper import Settings, get_hash, getUrl, ssplit
  File "C:\hostedtoolcache\windows\Python\3.9.13\x64\lib\site-packages\aqt\helper.py", line 650, in <module>
    def extract_auth(args: List[str]) -> Tuple[str | None, str | None, List[str] | None]:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
Error: Error: The process 'C:\hostedtoolcache\windows\Python\3.9.13\x64\python.exe' failed with exit code 1```

srcejon, 2025-04-23:
Looks like a problem with install-qt-action no longer supporting Python 3.9.

https://github.com/jurplel/install-qt-action/issues/287

f4exb, 2025-04-23:
Python 3.9 is now pretty old. I am wondering why this version gets installed and not a more recent one.

closed 2025-04-23 by f4exb

mattsoftware, 2025-04-24:
Perfect. Thank you.

commit a5d1f485 (2025-04-23): Try setup-python: true to fix #2441