f4exb/sdrangel #2446: 3D Mapn missing

KanotixPinguin, 2025-05-20:
Hello

For a few days now, the Earth has been missing on all my PCs; the grid isn't visible, and the 3D map is gone. It can't be my fault.

Greetings from Germany

srcejon, 2025-05-20:
We've exceeded our quota.

You can create your own Cesium Ion account here: https://cesium.com/platform/cesium-ion/ - create an Access Token and then enter it in Map > Display Settings > API Keys > Cesium Ion API Key

KanotixPinguin, 2025-05-20:
OK

Thanks

I thought so

KanotixPinguin, 2025-05-20:
I have a code here (https://ion.cesium.com/assets/2275207?page=1&sortBy=DATE_ADDED&sortOrder=DESC)

on the right side. Is that the one?

Or what exactly do I need to do?

srcejon, 2025-05-20:
Click Access Tokens at the top and create one. Copy the long string of characters in to Map > Display Settings > API Keys > Cesium Ion API Key in SDRangel

KanotixPinguin, 2025-05-20:
Thank you

It works

KanotixPinguin, 2025-05-20:
😎
[image]

srcejon, 2025-05-20:
I'm working on improving the lighting, and also adding a few other goodies 

[image]

KanotixPinguin, 2025-05-21:
I think that's great.
The program runs on two laptops for me, one AIS, the other ADBS.
Then in the main room, I have another laptop with TCP remote access for both, and I can see flight and marine on one map. Yes, the program is really expandable; you can do a lot with it and add a lot. I just have one problem. I'm still on Server 3 on the map in Karsruhe, Germany, and I can't get it off. Otherwise, it's a toy for parents, lol.

callumr39, 2025-08-30:
To further on from this it seems the map info section beside the logo seems to say “upgrade to commercial to use” any ideas it doesn’t seem anyone else has this issue.

srcejon, 2026-11-09:
Should be fixed in 8.1

closed 2026-11-09 by srcejon