f4exb/sdrangel #2451: Android build

wavesoft, 2025-06-01:
Hi all,

I have been using SDRAngel with LimeSDR Nano 2.0 for the last month and so far my experience is great. 

Unfortunately, when I tried the android version I didn't have the same luck. I noticed that the [official version on the store](https://play.google.com/store/apps/details?id=org.sdrangel&hl=en) is a bit old 7.21.3.

Therefore, I have 2 android related questions:

1. First, I couldn't find any instructions for an android build. I noticed that while it is technically possible to build SDRAngel binaries for the android, I couldn't find any android manifest, or other app-level launcher or parameters for the android app. Is the android build not part of the main project? Where can I find build instructions for it?
2. Second, let's say that I manage to build for android. My assumption would be that LimeSuite would be on a later version and therefore LimeSDR Nano 2.0 _would_ have been supported. Is my assumption correct, or are there any other reasons that blocks LimeSDR Nano 2.0 from being functional in the android environment?

prodikos, 2025-06-01:
I see... added my vote 👍 I guess the alternative would be linux in userspace + VNC Xvfb or something around these lines. I assume building for an arm64 CPU won't be of an issue 🤔 (but I guess getting Qt to install in that environment might be more of a challenge 😆 )

Do you know if there are any builds for any linux distribution that includes arm64 binaries?

wavesoft, 2025-06-01:
I see... added my vote 👍 I guess the alternative would be linux in userspace + VNC Xvfb or something around these lines. I assume building for an arm64 CPU won't be of an issue (but I guess getting Qt wizzard to install the libraries from the phone environment might be more of a challenge 😆 )

Do you know if there are any builds for any linux distribution that includes arm64 binaries?

prodikos, 2025-06-01:
Btw @srcejon , a few more questions, before I dig further in this rabbit hole:

1. Is there any known way to use LimeSDR Nano 2.0 with android? For example, is Soapy already included in the 7.21.3 android release? (and therefore the fact that it doesn't appear in the enumeration is caused by something else)
2. The workaround that I am trying is to launch `SoapyRemote` in Termux, using LimeSuite somehow interfacing through libusb with the LimeSDR device. And then using the remote source from SDRAngel to connect to the local server. Would that work?
3. Currently I am stuch, having trouble interfacing with libusb in userspace, so I am thinking of forking the [FX3 Connection](https://github.com/myriadrf/LimeSuite/tree/master/src/ConnectionFX3) and making it operate using an already open FD as the USB device (instead of using libusb). This would allow me to use more android-friendly APIs... which brings me to the final question. If you managed to build SDRAngel _with_ Soapy on android, how did you solve this libusb issue? 😆

srcejon, 2025-06-03:
Looks like it's built with LimeSuite 22.09.1. But yes, it needed some hacking to get it to work with Android USB and I only had the Lime SDR to try out. Think it only looks for VID/PID 1d50/6108

wavesoft, 2025-06-03:
Yeah from what I see android does not give you direct access to the device enumeration so `LimeSuite` won't work out-of-the box. At least in `termux` context it fails specaculously with a segfault 😆 

Following the principals of [termux-usb](https://wiki.termux.com/wiki/Termux-usb) that uses the Android API to retrieve a file descriptor of an already openned USB device, that we can then pass to my fork of LimeSuite, seems to do the trick. I have a fully functional connection. I even [exposed an fd property](https://github.com/wavesoft/LimeSuite/blob/6dd805c855b3aecccf0a325a4e36734204987a6b/src/ConnectionRegistry/ConnectionHandle.h#L73) on the connection handle, that you can use, if you already have such a handle obtained through the android API.

So, technically, if you do the enumeration in the app interface, obtain an FD for the USB device and then pass it to LimeSDR, it should work.

wavesoft, 2025-06-03:
FYI, with the previous approach, I got _somewhere_ but not completely where I wanted:

<table>
<tr>
<td>
[image]
</td>
</td>
</table>

DreamNik, 2025-06-04:
The android build is a complete mess because of dependency hell. Basically you need all the libraries to be available on build host (this is very unlikely) either integrate all the build scripts into project. Both of the variants are ugly.

I've went the first way and able to compiled SDRAngel v7.22.7 for myself.

Here is my apk:
https://drive.google.com/file/d/1LX8DSLdVeM_NzXk3p74PQwBLCaPeq9ly

The package is named "org.sdrangel2" so it can co-exists with original SDRAngel from store.

The source code is here:
https://github.com/DreamNik/sdrangel/tree/my
You can see that I basically ditched original dependencies and substituted my own.

That build is rather special. The things that broken:
- sdrplayv3 (not open source)
- map (requires QtWebEngine for android)
- screen repaint is buggy (Qt bug)
- device selection sometimes won't show USB device after permission request

All the other things should work if you device can handle it!
USB should work for all devices including but not limited to limesdr.