f4exb/sdrangel #2453: [Feature request] Noise filtering profiles

ElrikPiro, 2025-06-03:
# Context

For SDR (and other radio operating) users, noise seems to be a pretty common issue. Other solutions like the noise filtering plugin of SDR# allow users to filter different kind of audio tracks:

[image]

And I found it pretty useful, specially on bad signals setups.

# Proposal

To add a noise filtering stage that allows users to select noise filtering profiles that allow users to digitally remove noise for a more clear audio experience.

ElrikPiro, 2025-06-07:
Are you considering actioning this into the sdrangel project then?

ElrikPiro, 2025-06-10:
Thanks for taking the time for it

Please, keep me updated!

f4exb, 2026-01-04:
It looks like RNNoise source code is mirrored on Github: https://github.com/xiph/rnnoise