f4exb/sdrangel #2457: QT5 WEBENGINE Package on RPi

ibiglari, 2025-06-07:
`qt5-webengine-devel` is no longer available on Raspberry Pi 64bit based on bookworm. Compilation steps need to be updated to reflect this.

srcejon, 2025-06-09:
What are the changes needed? I don't have one to try.

ibiglari, 2025-06-10:
Quick change would be "SDRAngel does not compile on RPi anymore". I had a look into compiling QT5 Web Engine package on RPi from source, and it's not an easy feat.

srcejon, 2025-06-10:
Do you actually want to use the 3D Map? If not, you should be able to compile without Web Engine.

ibiglari, 2025-06-10:
I tried this:

```
cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DMIRISDR_DIR=/opt/install/libmirisdr \
-DAIRSPY_DIR=/opt/install/libairspy \
-DAIRSPYHF_DIR=/opt/install/libairspyhf \
-DBLADERF_DIR=/opt/install/libbladeRF \
-DHACKRF_DIR=/opt/install/libhackrf \
-DRTLSDR_DIR=/opt/install/librtlsdr \
-DLIMESUITE_DIR=/opt/install/LimeSuite \
-DIIO_DIR=/opt/install/libiio \
-DPERSEUS_DIR=/opt/install/libperseus \
-DXTRX_DIR=/opt/install/xtrx-images \
-DSOAPYSDR_DIR=/opt/install/SoapySDR \
-DUHD_DIR=/opt/install/uhd \
-DAPT_DIR=/opt/install/aptdec \
-DCM256CC_DIR=/opt/install/cm256cc \
-DDSDCC_DIR=/opt/install/dsdcc \
-DSERIALDV_DIR=/opt/install/serialdv \
-DMBE_DIR=/opt/install/mbelib \
-DCODEC2_DIR=/opt/install/codec2 \
-DSGP4_DIR=/opt/install/sgp4 \
-DLIBSIGMF_DIR=/opt/install/libsigmf \
-DDAB_DIR=/opt/install/libdab \
-DGGMORSE_DIR=/opt/install/ggmorse \
-DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel \
-DENABLE_FEATURE_MAP=OFF \
-DENABLE_FEATURE_SKYMAP=OFF ..
```

And I got the same error:

```
CMake Error at /usr/lib/arm-linux-gnueabihf/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5WebEngine" with
  any of the following names:

    Qt5WebEngineConfig.cmake
    qt5webengine-config.cmake

  Add the installation prefix of "Qt5WebEngine" to CMAKE_PREFIX_PATH or set
  "Qt5WebEngine_DIR" to a directory containing one of the above files.  If
  "Qt5WebEngine" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:722 (find_package)
```

f4exb, 2025-06-10:
In Ubuntu 24.04 you have `qtwebengine5-dev`. See the Docker file here: https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/armv8.ubuntu.Dockerfile#L69 

> SDRAngel does not compile on RPi anymore

Obviously wrong. It depends on the context.

f4exb, 2025-06-11:
Raspbian is not mentioned it says the reference environment for Linux is Ubuntu 22.04 which is wrong since it actually is 24.04 now which is consistent with what I said above.