f4exb/sdrangel #2470: sdrangelsrv remote TCP sink not working

SteveHawk, 2025-06-18:
Hey guys, I tried to setup a headless raspberry pi with `sdrangelsrv` running in docker with an RTL-SDR dongle, but could not get remote TCP sink to work.

The remote sink plugin using UDP does work, but when I switched to the TCP sink, server log shows a ton of overflow and dropped messages, and no waterfall shows up on the sdrangel client. Could you help me out here?

The client interface after connecting to the server:

[image]

The prefill buffer fills to 1s then back to 0 then to 1 and so on, but never starts showing the waterfall and spectrum.

The error logs on server:

```log
sdrangel     | 2025-06-18 10:47:00.680 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 3 messages dropped
sdrangel     | 2025-06-18 10:47:00.681 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 3072 samples
sdrangel     | 2025-06-18 10:47:00.922 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:03.437 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 217 messages dropped
sdrangel     | 2025-06-18 10:47:03.437 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:03.442 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:06.296 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 229 messages dropped
sdrangel     | 2025-06-18 10:47:06.296 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:06.300 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:08.803 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 229 messages dropped
sdrangel     | 2025-06-18 10:47:08.803 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:08.807 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:09.500 (D) RemoteTCPSinkSink::errorOccurred:  QAbstractSocket::RemoteHostClosedError
sdrangel     | 2025-06-18 10:47:09.500 (D) RemoteTCPSinkSink::disconnected
sdrangel     | 2025-06-18 10:47:20.919 (D) RemoteTCPSinkSink::acceptTCPConnection: client connected
sdrangel     | 2025-06-18 10:47:22.209 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 77 messages dropped
sdrangel     | 2025-06-18 10:47:22.209 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 3072 samples
sdrangel     | 2025-06-18 10:47:22.213 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:25.022 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 219 messages dropped
sdrangel     | 2025-06-18 10:47:25.023 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 3072 samples
sdrangel     | 2025-06-18 10:47:25.027 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:27.828 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 216 messages dropped
sdrangel     | 2025-06-18 10:47:27.828 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 3072 samples
sdrangel     | 2025-06-18 10:47:27.833 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:30.638 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 218 messages dropped
sdrangel     | 2025-06-18 10:47:30.638 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 3072 samples
sdrangel     | 2025-06-18 10:47:30.642 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:33.448 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) 218 messages dropped
sdrangel     | 2025-06-18 10:47:33.448 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 3072 samples
sdrangel     | 2025-06-18 10:47:33.453 (C) SampleSinkFifo::write: (RemoteTCPSink [0:0]) overflow - dropping 4096 samples
sdrangel     | 2025-06-18 10:47:35.288 (D) RemoteTCPSinkSink::errorOccurred:  QAbstractSocket::RemoteHostClosedError
sdrangel     | 2025-06-18 10:47:35.288 (D) RemoteTCPSinkSink::disconnected
```

This is the startup command I used:
`/opt/install/sdrangel/bin/sdrangelsrv -w /home/sdr/.config/sdrangel/fftw-wisdom --remote-tcp --remote-tcp-hwtype RTLSDR`

And this is the complete compose.yaml:

```yaml
---
services:
  sdrangel:
    image: "f4exb06/sdrangelsrv:v7.22.7"
    container_name: "sdrangel"
    init: true
    restart: unless-stopped
    ports:
      - "8091:8091"  # API
      - "1234:1234"  # remote tcp
      # - "8887:8887"  # spectrum
      # - "9090:9090/udp"  # remote input udp port
    devices:
      - "/dev/bus/usb:/dev/bus/usb"
    environment:
      FFTWFILE: "fftw-wisdom"
      LD_LIBRARY_PATH: "/opt/install/libsdrplay/lib:/opt/install/xtrx-images/lib"
    volumes:
      - "/home/steve/Applications/services/sdrangel/configs:/home/sdr/.config:rw"
    entrypoint:
      - /opt/install/sdrangel/bin/sdrangelsrv
      - -w
      - /home/sdr/.config/sdrangel/fftw-wisdom
      - --remote-tcp
      - --remote-tcp-hwtype
      - RTLSDR

  sdrangelcli:
    image: "f4exb06/sdrangelcli:v4.0.2"
    container_name: "sdrangelcli"
    init: true
    restart: unless-stopped
    ports:
      - "8080:8080"
```

I did also try running `rtl_tcp` and `spyserver` on the raspi, both work with sdrangel client using default settings. So it appears the client should be working fine, but the server side `sdrangelsrv` for some reason is failing to send out samples?

srcejon, 2025-06-18:
Can you try with sdrangel as the server, rather than sdrangelsrv?

srcejon, 2025-06-19:
> First, the `--remote-tcp` flag doesn't seem to work on `sdrangel`, the UI shows up with an empty workspace, so I have to manually add the device and remote tcp sink.

I'm only getting a FileSink appear. It used to work. I guess I broke it.

>  Then I figured out it's the default 2M/s SR being too high for my setup. I need to lower the channel sample rate before connect. 

What network connection do you have?

> That's it! Turns out I'm a dummy and don't know how to tune the parameters. It's SR and Ch gain all along. But here's the question: why does `sdrangel` and `sdrangelsrv` behave differently in this case?

It shouldn't, so sounds like there's a couple of bugs.

SteveHawk, 2025-06-19:
> What network connection do you have?

I'm using wifi on both sides. When using sdrangel, the max stable SR is about 1.5MS/s (around 500KB/s), but using rtl_tcp I can push to around 2.8MS/s (around 6MB/s).

I think my bottleneck is the CPU here instead of network. rtl_tcp doesn't do compression so not much cpu usage at all. For sdrangel, at 1.5MS/s, it uses 1 cpu core at about 95%. When push to 1.6-1.7, I can see that core usage reaching 100%, and overflow logs start showing up.

SteveHawk, 2025-06-19:
Tried all three options, `IQ only` doesn't help if not worse, FLAC/zlib level 0 is only slightly better, I pushed to about 1.9MS/s before fifo overflowing. Still blocked on that weak weak single core performance.

On a side note, `sdrangelcli` doesn't support remote TCP sink's config at the moment. Is it possible to implement the option in the future?

srcejon, 2025-06-19:
Is there channel sample rate different to the device sample rate? For lowest CPU usage, lock those to be the same and just adjust the device sample rate.

srcejon, 2025-06-19:
And then it goes from 20% to over 100% when a client connects with IQ only? Strange.

Any chance of a screenshot of the server when the client is connected please?

I should have a fix for starting from the cli shortly./

srcejon, 2025-06-19:
Thanks.

First thing to note, you should have the device "gain" at max, not the "channel gain". You can see the RTL SDR's gain is set to 0 currently, which means you wont any signal. The "Ch gain" is just if you are trying to amplify a small 12-bit SDR signal in to 8-bits.

SteveHawk, 2025-06-19:
> you should have the device "gain" at max, not the "channel gain"

Yup I know, thanks for the heads-up. My client was connected to the sdrangelsrv, that's why the channel gain was cranked. I realized that after taking the first screenshot lol, you can see the second screenshot I have turned down the channel gain.

> The "Ch gain" is just if you are trying to amplify a small 12-bit SDR signal in to 8-bits.

So is this related to the bug in sdrangelsrv? Like it's not properly converting a 12-bit signal to 8-bit?

srcejon, 2025-06-23:
> So my point is that there are some inconsistencies about ch gain between sdrangel and sdrangelsrv.

Ok. There shouldn't be any difference though as it should be the exact same code.

In both cases are you running the server and client on the same OS & hw? Was the compiler the same for both?

Can you run:

    lscpu | grep Endian

Just to check the endianness of your CPU.

SteveHawk, 2025-06-23:
Yes, server and client are on the same OS & hw respectively.

Client:

An x86 laptop running Linux (Pop!_OS 22.04). `sdrangel` client is the flatpak version 7.22.7.

```bash
> lscpu | grep Endian
Byte Order:                           Little Endian
```

Server:

A raspberry pi 4B running raspberry pi os bookworm. 

```bash
> lscpu | grep Endian
Byte Order:                           Little Endian
```

Both `sdrangel` and `sdrangelsrv` are running inside their docker container, base distro are both Ubuntu 24.04. `sdrangel` is built from this Dockerfile: <https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/armv8.ubuntu.Dockerfile> and `sdrangelsrv` image is pulled from here: <https://hub.docker.com/r/f4exb06/sdrangelsrv/tags> (f4exb06/sdrangelsrv:v7.22.7)

Now i'm realizing the `sdrangel` is not really built using the exact same code as `sdrangelsrv`, because the `build_vanilla.sh` script defaults to use master branch. Some commits in the last two months changed the behavior maybe? I'll try build `sdrangel` on tag v7.22.7 and see if anything changes.

srcejon, 2025-06-23:
> Some commits in the last two months changed the behavior maybe?

Not that I can think of.

SteveHawk, 2025-06-23:
Ok tried recompile on tag v7.22.7, nothing changed.

srcejon, 2025-06-23:
`sdrangel` is built from this Dockerfile: https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/armv8.ubuntu.Dockerfile and `sdrangelsrv` image is pulled from here: https://hub.docker.com/r/f4exb06/sdrangelsrv/tags (f4exb06/sdrangelsrv:v7.22.7)

Ah, that says "SDRangel server mode with 16 bit Rx samples in arm64"

I've only ever tested it with 24 bit samples (-DRX_SAMPLE_24BIT=ON) - so there could well be a bug there.

SteveHawk, 2025-06-23:
Ah that's interesting, I can confirm from the log that the `sdrangelsrv` is using `Rx:16b` and vanilla gui `sdrangel` is using `Rx:24b`.

srcejon, 2025-06-23:
Ok, there was no code for the 16-bit case. Let me know how you get on with the above patch.