f4exb/sdrangel #2485: HydraSDR

henrikkoch, 2025-07-13:
Any support for HydraSDR planned?

alphafox02, 2025-07-21:
Surely it does, but if hydras driver source was added it would be good from the perspective of not hijacking Airspys driver for the HydraSDR. I’m not sure why HydraSDR One has the same udev info.

srcejon, 2025-07-22:
> if hydras driver source was added 

I'm not sure there's any source to add.

> I’m not sure why HydraSDR One has the same udev info.

Presumably so that it is compatible with any existing software that already supports the Airspy.

alphafox02, 2025-07-22:
True, but it’s also been mentioned there’s ongoing litigation regarding just that. Knowing this, I just stick to promoting the use of it with its own sources from the HydraSDR repository.

alphafox02, 2025-07-22:
And having glanced just now, it would appear they’re taking steps to rectify things 

“ New Official: USB.org registered VID 0x38af / PID 0x0001”

srcejon, 2025-07-24:
Yes. Looks like you'd just need to copy and paste the airspy plugin changing airspy_ to hydrasdr_