f4exb/sdrangel #2487: Program freezes and/or closes randomly

barclaystation, 2025-07-18:
After ADSB feature update, SDRAngel freezes or spontaneously closes after random periods of time, usually a few hours or less. With previous version, program could be left running for days to track and log aircraft without issue. Using RTLSDR V4.

barclaystation, 2025-07-18:
Also, I forgot to specify that I am running the program on Windows 11.

srcejon, 2025-07-19:
Please attach a screenshot showing your setup, thanks.

barclaystation, 2025-07-19:
Thank you, I appreciate your time, and it is a fantastic program. I run it with two displays, screenshots attached. 

[image]

[image]

srcejon, 2025-07-22:
Just run here for 24 hours with no problems or signs of a memory leak.

Perhaps you could attach the adsb log file you have created after it crashes? Also the debug log file (Preferences > Logging). Thanks.

Maybe a screenshot if it hangs again?

barclaystation, 2025-07-23:
Good morning, it froze again sometime last night, attached are the log files and screenshots. Thank you much

[ADSB LOG MAIN.csv](https://github.com/user-attachments/files/21388391/ADSB.LOG.MAIN.csv)

[sdrangel.log](https://github.com/user-attachments/files/21388448/sdrangel.log)

[image]
[image]

srcejon, 2025-07-23:
Thanks - can't see anything obviously wrong in the logs, unfortunately.

barclaystation, 2025-07-31:
Great, thank you. I've got it loaded and running now, and will report back.

barclaystation, 2025-08-10:
I have finally noticed a correlation when the program freezes. The last few times I observed the program to freeze, I have noticed that the debugging log shows the last line item occurrence at the moment which I wake the screen and unlock the computer.  

[sdrangel.log](https://github.com/user-attachments/files/21703428/sdrangel.log)

srcejon, 2025-08-10:
There's nothing in the log.

I've tried locking my screen several times and it doesn't seem to be an issue.

You could perhaps try locking/unlocking repeatedly to see if you can cause it freeze. Or disable locking to see if it then runs OK.

closed 2025-09-25 by github-actions[bot]