f4exb/sdrangel #2489: Wayland does not support QWindow

MoldyTaint, 2025-07-27:
[sdrangel.log](https://github.com/user-attachments/files/21450728/sdrangel.log)

srcejon, 2025-07-29:
Aside from the warning, is something not working?

srcejon, 2025-07-29:
Which Map? 2D, 3D, ADS-B? Screenshot might help.

MoldyTaint, 2025-07-29:
It would be the ADS-B for planes and such. 

System Details are here as well: 

Operating System: CachyOS Linux 
KDE Plasma Version: 6.4.3
KDE Frameworks Version: 6.16.0
Qt Version: 6.9.1
Kernel Version: 6.15.6-2-cachyos (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 7800X3D 8-Core Processor
Memory: 32 GiB of RAM (30.5 GiB usable)
Graphics Processor 1: AMD Radeon RX 7800 XT
Graphics Processor 2: AMD Radeon Graphics
Manufacturer: Gigabyte Technology Co., Ltd.
Product Name: B650 AORUS ELITE AX