f4exb/sdrangel #2494: ATV demodulator for a headless machine.

NikBarykin, 2025-08-03:
Hey there! 
I have a headless server and I really want to demodulate a tv signal. However I do not encounter an ATV demodulator in sdrangelcli. How can I achieve this goal? Is there some workarounds?

NikBarykin, 2025-08-04:
Evening!

Why do you think that there is no much use of ATV? My goal is to capture a signal on a headless machine decode it in the video and transfer the video to the main machine.  Is there some workaround maybe like decoding the video on the main machine? 

The thing is that I want to control the whole process from my Python script. And I cannot help finding TV signal decoder.

Advance thanks!

NikBarykin, 2025-08-07:
I think there is some misunderstanding.

One way or another original SDRAngel is able to capture ATV signal and produce a video on my screen. I want to perform similar thing: to receive a signal on a remote server and display an image on my main computer (and to save the whole video somewhere). There is no way that it is impossible.

f4exb, 2025-08-07:
> receive a signal on a remote server 

Do you mean just receive the I/Q stream via the network and perform the demodulation on your main computer?

f4exb, 2025-08-07:
Then at the remote end (headless machine) you may want to use this plugin to send the I/Q samples over the network: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/remotetcpsink

NikBarykin, 2025-08-07:
That’s nice. However the main problem remains: how to call a demodulation from script?

f4exb, 2025-08-07:
I understood that you would like to handle the demodulation with your own software written in Python.