f4exb/sdrangel #2495: DAB-demodulator broken

gvanem, 2025-08-03:
Since Jan's huge commit here:   https://github.com/JvanKatwijk/dab-cmdline/commit/9d6f45f6f7e48355c31acae7a613e9d0f1de2a29

this has broken the build of `channelrx/demoddab` in SDRangel. The former C-API in `dab-api.h` now uses `std::string` etc. etc.
So `channelrx/demoddab` no longer compiles. 

Some of the errors I'm getting:
```c
f:\gv\dx-radio\DAB\DAB-cmdline\dab-api.h(258): warning C4190: 'dab_getserviceName' has C-linkage specified, but returns 'std::basic_string<c
har,std::char_traits<char>,std::allocator<char>>' which is incompatible with C (compiling source file DABdemod.cpp)
f:\gv\VC_2022\VC\Tools\MSVC\14.44.35207\include\xstring(541): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::a
llocator<char>>' (compiling source file DABdemod.cpp)
f:\gv\dx-radio\DAB\DAB-cmdline\dab-api.h(258): warning C4190: 'dab_getserviceName' has C-linkage specified, but returns 'std::basic_string<c
har,std::char_traits<char>,std::allocator<char>>' which is incompatible with C (compiling source file objects/moc_DABdemodBaseband.cpp)
f:\gv\VC_2022\VC\Tools\MSVC\14.44.35207\include\xstring(541): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::a
llocator<char>>' (compiling source file objects/moc_DABdemodBaseband.cpp)
f:\gv\dx-radio\DAB\DAB-cmdline\dab-api.h(258): warning C4190: 'dab_getserviceName' has C-linkage specified, but returns 'std::basic_string<c
har,std::char_traits<char>,std::allocator<char>>' which is incompatible with C (compiling source file DABdemodDevice.cpp)
f:\gv\VC_2022\VC\Tools\MSVC\14.44.35207\include\xstring(541): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::a
llocator<char>>' (compiling source file DABdemodDevice.cpp)
f:\gv\dx-radio\DAB\DAB-cmdline\dab-api.h(258): warning C4190: 'dab_getserviceName' has C-linkage specified, but returns 'std::basic_string<c
har,std::char_traits<char>,std::allocator<char>>' which is incompatible with C (compiling source file DABdemodBaseband.cpp)
f:\gv\VC_2022\VC\Tools\MSVC\14.44.35207\include\xstring(541): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::a
llocator<char>>' (compiling source file DABdemodBaseband.cpp)
f:\gv\dx-radio\DAB\DAB-cmdline\dab-api.h(258): warning C4190: 'dab_getserviceName' has C-linkage specified, but returns 'std::basic_string<c
har,std::char_traits<char>,std::allocator<char>>' which is incompatible with C (compiling source file DABdemodWebApiAdapter.cpp)
f:\gv\VC_2022\VC\Tools\MSVC\14.44.35207\include\xstring(541): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::a
llocator<char>>' (compiling source file DABdemodWebApiAdapter.cpp)
f:\gv\dx-radio\DAB\DAB-cmdline\dab-api.h(258): warning C4190: 'dab_getserviceName' has C-linkage specified, but returns 'std::basic_string<c
har,std::char_traits<char>,std::allocator<char>>' which is incompatible with C (compiling source file DABdemodPlugin.cpp)
f:\gv\VC_2022\VC\Tools\MSVC\14.44.35207\include\xstring(541): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::a
llocator<char>>' (compiling source file DABdemodPlugin.cpp)
f:\gv\dx-radio\DAB\DAB-cmdline\dab-api.h(258): warning C4190: 'dab_getserviceName' has C-linkage specified, but returns 'std::basic_string<c
har,std::char_traits<char>,std::allocator<char>>' which is incompatible with C (compiling source file DABdemodSink.cpp)
f:\gv\VC_2022\VC\Tools\MSVC\14.44.35207\include\xstring(541): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::a
llocator<char>>' (compiling source file DABdemodSink.cpp)
DABdemodSink.cpp(516): error C2039: 'ensemblename_Handler': is not a member of 'API_struct'
f:\gv\dx-radio\DAB\DAB-cmdline\dab-api.h(179): note: see declaration of 'API_struct'
```

srcejon, 2025-08-04:
Instructions are to build with:

    git clone https://github.com/srcejon/dab-cmdline
    cd dab-cmdline/library
    git checkout msvc

gvanem, 2025-08-04:
> git clone https://github.com/srcejon/dab-cmdline

Hey, that works fine. I even get good and clear sound. More than I can say about Jan's version.

closed 2025-08-04 by gvanem