f4exb/sdrangel #2501: Not able to move AIS window into a different (own) Workspace

mjanicki, 2025-08-25:
Tried to move the AIS window into it's own workspace. The AIS demodulator window(s) will move.

A somewhat related Question, is it possible to record just 1 or selected AIS targets to save to a file?

Thanks,
Rgds,
   Mike J

srcejon, 2025-09-01:
> Tried to move the AIS window into it's own workspace. The AIS demodulator window(s) will move.

How exactly did you try? I just gave it a try via the Move to another workspace button and it worked.

> A somewhat related Question, is it possible to record just 1 or selected AIS targets to save to a file?

Not currently.

srcejon, 2025-09-01:
Ok, the problem seems to apply to all Features, not just AIS, but it isn't always a problem.

When you first open a feature, then the Move to another workspace dialog works OK.
However, if you have features open and then restart SDRangel so the features are restored from the default config, then the Move to another workspace dialog does not appear to work.

srcejon, 2025-10-04:
Fixed in 7.22.9