f4exb/sdrangel #2520: Build failed with latest libdab

CoelacanthusHex, 2025-09-20:
```
/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp: In constructor ‘DABDemodSink::DABDemodSink()’:
/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp:516:11: error: ‘struct API_struct’ has no member named ‘ensemblename_Handler’
  516 |     m_api.ensemblename_Handler = ensembleNameHandler;
      |           ^~~~~~~~~~~~~~~~~~~~
/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp:517:11: error: ‘struct API_struct’ has no member named ‘programname_Handler’; did you mean ‘programdata_Handler’?
  517 |     m_api.programname_Handler = programNameHandler;
      |           ^~~~~~~~~~~~~~~~~~~
      |           programdata_Handler
/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp:525:30: warning: invalid conversion from ‘void (*)(int, void*)’ to ‘tii_data_t’ {aka ‘void (*)(tiiData*, void*)’} [-fpermissive]
  525 |     m_api.tii_data_Handler = tiiDataHandler;
      |                              ^~~~~~~~~~~~~~
      |                              |
      |                              void (*)(int, void*)
/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp: In member function ‘void DABDemodSink::setProgram(const QString&)’:
/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp:659:45: error: invalid initialization of non-const reference of type ‘audiodata&’ from an rvalue of type ‘audiodata*’
  659 |         dataforAudioService(m_dab, program, &m_ad, 0);
      |                                             ^~~~~
In file included from /build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemoddevice.h:24,
                 from /build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.h:33,
                 from /build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodbaseband.h:30,
                 from /build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemod.h:31,
                 from /build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp:27:
/usr/include/dab_lib/dab-api.h:236:53: note: in passing argument 3 of ‘void dataforAudioService(void*, const std::string&, audiodata&, int)’
  236 |                                                     audiodata &, int);
      |                                                     ^~~~~~~~~~~
/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp:667:37: error: invalid initialization of non-const reference of type ‘audiodata&’ from an rvalue of type ‘audiodata*’
  667 |             set_audioChannel(m_dab, &m_ad);
      |                                     ^~~~~
/usr/include/dab_lib/dab-api.h:247:50: note: in passing argument 2 of ‘void set_audioChannel(void*, audiodata&)’
  247 | void DAB_API    set_audioChannel        (void *, audiodata &);
      |                                                  ^~~~~~~~~~~
```
libdab commit: https://github.com/JvanKatwijk/dab-cmdline/commit/ca810ec564174cc9bdf5e108b42ba858665fa758
sdrangle commit: https://github.com/f4exb/sdrangel/commit/bf4d4a1b782dea1fe81d25489bcf6c0e8c5cc01b
Use Arch Linux, GCC 15.2.1

srcejon, 2025-09-21:
[Instructions](https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux) are to build with:

    git clone https://github.com/srcejon/dab-cmdline
    cd dab-cmdline/library
    git checkout msvc

srcejon, 2026-10-30:
Closing, as we no longer use it.

closed 2026-10-30 by srcejon